brainstorm-tools/brainstorm3 #308: Internet Connection

gsdouk, 2020-06-14:
Hello,

I encounter a problem connecting to server via BST, cannot download updates etc. When BST initializes, the internet connectivity is fine (Checking internet connectivity... ok). I did not had this problem before the 02/June update. 

Thank you in advance.

ftadel, 2020-06-15:
Can you please describe better you issue(s)?
What is happening? When? 
Please copy-paste the full error messages you get, including the stack traces.

gsdouk, 2020-06-15:
On BST's initialization I get: 
[image]

The error I get when I am trying to update BST:
[image]

One more example (similar with the above), when I am trying to compute MNI transformation on the MriViewer when importing anatomy: 
[image]

The stack traces:
[image]

The 15/June update did not fix this error. Before the 02/June update I did not have any problem updating BST.

ftadel, 2020-06-15:
Do you have an older version of Brainstorm available somewhere?
Could you run it and try to run the update from it?

We're having certificate issues on the webserver at the moment, this might be related with this problem. If this is the cause, any older version of Brainstorm should not work any better.

What do you get if you execute the line below in your Matlab command window?
`webread('http://neuroimage.usc.edu/bst/getversion_duneuro.php')`

While we're working on this issue, please update Brainstorm manually (download a new version from the website) and/or use the compiled version.

gsdouk, 2020-06-15:
>Do you have an older version of Brainstorm available somewhere?
Could you run it and try to run the update from it?

Yes, I run an older version (23/Sep/19) and I still get the same error message. 

>What do you get if you execute the line below in your Matlab command window?
webread('http://neuroimage.usc.edu/bst/getversion_duneuro.php')

[image]

>While we're working on this issue, please update Brainstorm manually (download a new version from the website) and/or use the compiled version.

Actually, I do not care about the updates (since I manually updated BST on the last versions). What I am interested about are the anatomy templates and atlases I should manually download and import on BST .

ftadel, 2020-06-16:
For the MNI template used for the SPM registration with SPM, download it from:
https://neuroimage.usc.edu/bst/getupdate.php?t=SPM_TPM
And unzip it in your $HOME/.brainstorm/defaults/spm/ folder (full path indicated in one of your screen captures)

Which other files are failing to get downloaded automatically?

Hopefully, this will get fixed quickly...

ftadel, 2020-06-18:
After looking into the SSL certification chain, I understood that the problem is not related with USC, but with the expiration of a major root certificate on May 30th.
https://www.sslshopper.com/ssl-checker.html#hostname=neuroimage.usc.edu
https://www.xolphin.com/support/Rootcertificates/Phasing_out_Addtrust_External_CA_Root_certificate

We solved part of the issue by:
- Removing the automatic redirection from HTTP to HTTPS for our update PHP scripts 
- Switching to HTTP all the URLs used by Brainstorm to download updates, templates and additional modules: https://github.com/brainstorm-tools/brainstorm3/commit/5fccdb4e7e2f5990583dd8aa049a61844f8889f3
This will solve the problem in a few cases, but the automatic updates might keep on failing for a large number of Brainstorm users.

**Workaround**: 
Simply delete your current version of Brainstorm (the folder "brainstorm3" with the program) and and downloading a new one from the website: https://neuroimage.usc.edu/bst/download.php

gsdouk, 2020-06-18:
I installed the latest version (18/June) and everything seems to work just fine. 

Thank you very much for fixing this bug.

closed 2020-06-18 by ftadel