brainstorm-tools/brainstorm3 #310: Error while running DUNeuro FEM

gsdouk, 2020-06-20:
Hello,

About the Realistic head model: FEM mesh generation, I have read on the [tutorial](https://neuroimage.usc.edu/brainstorm/Tutorials/FemMesh) that it is under development but it is not clear to me if the method or the tutorial is under development. 

In any case, I have tried to run this method and the stack trace I get is:
[image]
The unfinished sentence says: _No such file or directory_.

I have installed the libc6 package on an Ubuntu distribution and I use the 20/June BST's version.

tmedani, 2020-06-20:
Hi @gsdouk 

Thanks for reporting this issue. 

Both are under development, the code, and the tutorial. 

Regarding your error, this is already reported here : 
https://neuroimage.usc.edu/forums/t/duneuro-error-libdl-so-2/19203/8

Which OS are you using?

The binaries are tested on Ubuntu.  

@juangpc any updates regarding this problem?

juangpc, 2020-06-22:
Hi, 

These are system-level libraries. The code depends on them, and sorry for the frustration, but really it depends on the operating system, on whether you admit 3rd party SW or not, licensing, etc... And we can't test all different flavors. We've tested Ubuntu lts and it works fine. 

At some point in the future, I'll try to build a static version. 

I would recommend waiting until the feature is fully finished and tested.

gsdouk, 2020-06-22:
Hello,

The system I use runs on Ubuntu 16.04 LTS. Thank you for the information.

closed 2020-06-22 by gsdouk