brainstorm-tools/brainstorm3 #314: Duneuro installation check in mac platform

juangpc, 2020-07-09:
Hello, 

There seems to be a problem with matlab's ```which``` command in mac whenever the file has no extension.

Right now, even when the duneuro toolbox is correctly installed, the following ```which``` outputs empty (no path) and the script assumes the toolbox was not properly installed.

This problem does not happen in windows, nor whenever the binary file's name has an extension.

Right now, I've found a solution which is to rename the duneuro binary by adding a ".app" suffix. We would add that suffix to both mac and linux environments. There should be no other effect.

https://github.com/brainstorm-tools/brainstorm3/blob/935000ffea4e2b62c1053e8176e944ee68b4e324/toolbox/forward/duneuro_install.m#L40

tmedani, 2020-07-09:
Hi, 

I think also that is a fast way to solve the problem. 

By the way, I think also that the bash file need to be updated for future compilation.

https://github.com/brainstorm-tools/bst-duneuro/blob/82512c344925013dd3433d912f9ec70700b68dcd/config/setup_bst_duneuro.sh#L170

juangpc, 2020-07-09:
Thanks, I've created a new branch there. So, in case everybody is ok with the changes, we just have to merge it into master. 

https://github.com/brainstorm-tools/bst-duneuro/tree/app_extension_mac_linux

ftadel, 2020-07-13:
There is maybe a simpler option for finding the file using something else than the Matlab's which on MacOS.
Like using the system "which" from an instance of a shell on which we set the Matlab path?

But if you already have it figured out completely for both bst-duneuro and Matlab using an extra ".app", you can go ahead and merge.

juangpc, 2020-07-13:
Ok. Will do. 

But let me just be sure of this. If I rename the binary files in the bin/ folder in bst_duneuro this way... 

```bin/bst_duneuro_meeg_linux64 → bin/bst_duneuro_meeg_linux64.app```

```bin/bst_duneuro_meeg_mac64 → bin/bst_duneuro_meeg_mac64.app```

will they be correctly grabbed and synced with our sever?

ftadel, 2020-07-14:
Good point!

I checked the update script: it exports from the git repo all the files in https://github.com/brainstorm-tools/bst-duneuro/tree/master/bin.
Then it makes all the files executables with: `chmod a+x ${tmpdir_current}/bst_duneuro/bin/bst_duneuro*`

So the packaging script should not be affected by adding `.app` to the file names.

juangpc, 2020-07-14:
Thanks. 

As of the following commit, update running production now. 

https://github.com/brainstorm-tools/bst-duneuro/commit/57bf99a22ddf1862d07ffcba86bdf702afb48fe2

I'll close this issue.

closed 2020-07-14 by juangpc