brainstorm-tools/brainstorm3 #320: Brainstorm-Duneuro FEM tutorial

tmedani, 2020-08-10:
Hi, 

I was hesitating if we need to discuss it here or on the forum, so finally, it's here, 

@ftadel I would like to complete the tutorial  https://neuroimage.usc.edu/brainstorm/Tutorials/Duneuro#preview

but since you have already added some inputs, I don't want to remove or change it. 

we need to agree for a plan and then complete it. 

Regarding the data, we have the full EEG/MEG, we have the FEM head model (the mesh and the tensor) that are open source, however, we can not make the MR data available :(

So, we can use only the resulting mesh and combine it with the EEG/MEG data for the tutorial

To show the users how to generate a head model,  we may need an independent full dataset (T1, T2 and dwi) and maybe an independent small tutorial dedicated to that.

Does this make sense for you?

A+

ftadel, 2020-08-14:
In practice, the two sets of tools (mesh generation and FEM computation) will not be used separately by our users. These are two steps of the same FEM pipeline, which should be therefore be two parts of the same tutorial.
We would maximize the benefits of this pipeline if it were documented as one single tutorial, which would also match the published article.

Does this mean that in the article you are writing, you won't be able to provide the dataset either? 

@jcmosher @juangpc 
Didn't you have full datasets that we could use as well? (T1+T2+DWI+MEG and/or EEG)

juangpc, 2020-08-14:
We have T1+T2+MEG+EEG but not DWI, of the same patient. When we talked about this whole thing the diffusion part was still on the horizon. But time goes fast...

ftadel, 2020-08-15:
No hope for getting a full dataset?

What's the roadblock with the initial dataset you want wanted to use?

tmedani, 2020-08-17:
> No hope for getting a full dataset?

Less chance, it can be possible, we have to check  on humanconnectome.org and openneuro.org

> What's the roadblock with the initial dataset you want wanted to use?

It was a misunderstanding regarding between using or sharing the full data, 
the owner of the data agreed to use internally the MR data but not for public distribution. 

The good point is that we can distribute the FEM mesh with the EEG/MEG data.

ftadel, 2020-08-18:
Then maybe we could organize the tutorials differently:
1) We write one complete tutorial for the T1+T2+MEG+EEG from John/Juan, that covers both the FEM mesh generation and the duneuro computation (isotropic case only), starting from raw files as real users would have. This would be the standard use case for most Brainstorm users interested in using FEM models, right?
2) We write a second, more advanced tutorial illustrating the DWI/anistotropy pipeline, that starts from the FEM mesh already imported. This could be distributed as a Brainstorm protocol to load.

Would that work for you?

juangpc, 2020-08-18:
That could be a good compromise solution.

tmedani, 2020-08-18:
Hello,

I agree for 1,
however, for 2, I think we will still need the T1 data and the DWI in order to use illustrate the full pipeline.

ftadel, 2020-08-18:
> however, for 2, I think we will still need the T1 data and the DWI in order to illustrate the full pipeline.

But you said you can't distribute them...
Or am I missing something?

tmedani, 2020-08-18:
> > however, for 2, I think we will still need the T1 data and the DWI in order to illustrate the full pipeline.
> 
> But you said you can't distribute them...
> Or am I missing something?

sorry, it was not clear

yes, we can't distribute them. 

and in order to illustrate the DWI/anistotropy pipeline, we still need the T1 and DWI, and these are not available right now.

ftadel, 2020-08-18:
Easy then: no DWI/anisotropy tutorial.
Only one online tutorial with John/Juan's data.

In the SPIE article, are you using the data you can't distribute? This would mean:
1) you won't distribute your data with your article
2) your results won't be reproducible 
3) it won't match any online Brainstorm tutorial

Shall we have a discussion about good practice for open science at the next meeting? :)

tmedani, 2020-08-18:
> Easy then: no DWI/anisotropy tutorial.

My idea is to have a small tutorial independent from the forward computation,
where we can show how to use the DWI/tensors pipeline. 

I have some data with T1/T2 and DWI from a same subject that can be used for this purpose (without EEG/MEG)
==> does this work for you?

We can refer to this tutorial from the main tutorial with juan/john data. 

> In the SPIE article, are you using the data you can't distribute? This would mean:

True, what we can do is just a small page which will be  linked to this abstract, where we will distribute the mesh and the link to the EEG/MEG dataset, and link to another tuto to show how to generate the mesh. 

So we will have a set of small tutorials for each topic, 
- one for FEM mesh generation from T1 and T2, then source localization from EEG/MEG with Juan/John's data (isotropic model)
- one small tutorial for the mesh generation and DWI/aniso pipeline with only  T1, T2 and DWI (SimNibs Data)
- one samll tuto for the SPIE paper and maybe the full paper (distribute the mesh and link to the EEG/MEG). 

All theses tutos will be linked together. 

==> does all this make sense?

> Shall we have a discussion about good practice for open science at the next meeting? :)

it always good to discuss these points :)

ftadel, 2020-08-19:
> one for FEM mesh generation from T1 and T2, then source localization from EEG/MEG with Juan/John's data (isotropic model)

Yes.

> one small tutorial for the mesh generation and DWI/aniso pipeline with only T1, T2 and DWI (SimNibs Data)

Yes.

> one samll tuto for the SPIE paper and maybe the full paper (distribute the mesh and link to the EEG/MEG).

Not sure, I don't see what would be the added value for it.
1) It would not allow to replicate your findings, because it would start only from fully processed anatomical data.
2) It would not have any teaching value, as all the FEM algorithms and elements of interface in Brainstorm are already introduced in the first two tutorials
3) It seems to me it would only make the maintenance of the documentation heavier. Redundancy in software documentation should be avoid exactly as in software code (eg. if we update the process options, we want to have to redo the screen captures and edit the text only in one page if possible, not it two)

tmedani, 2020-08-19:
> > one for FEM mesh generation from T1 and T2, then source localization from EEG/MEG with Juan/John's data (isotropic model)
> 
> Yes.

Ok, we will work on that with @juangpc 

> > one small tutorial for the mesh generation and DWI/aniso pipeline with only T1, T2 and DWI (SimNibs Data)
> 
> Yes.

I will process the data and then create a small tuto

> > one samll tuto for the SPIE paper and maybe the full paper (distribute the mesh and link to the EEG/MEG).
> 
> Not sure, I don't see what would be the added value for it.

ok, make sense, referring to the two previous should be sufficient.

ftadel, 2021-09-19:
I guess we don't need this issue anymore.

closed 2021-09-19 by ftadel