brainstorm-tools/brainstorm3 #326: new error with BDP with Ernie dataset (SimNibs data)

tmedani, 2020-08-19:
Hello, 

This thread reports an error with BDP while applying it to the SimNIbs dataset example (Ernie). 

with @ajoshiusc, we are going to investigate the problem, and this discussion will track the solution 
that could be helpful for future users. 

here is the output :

BST> System call: bse -i "C:\Users\33649\.brainstorm\tmp\brainsuite\ErnieT1.nii" --auto -o "C:\Users\33649\.brainstorm\tmp\brainsuite\skull_stripped_mri.nii.gz" --mask "C:\Users\33649\.brainstorm\tmp\brainsuite\bse_smooth_brain.mask.nii.gz" --hires "C:\Users\33649\.brainstorm\tmp\brainsuite\bse_detailled_brain.mask.nii.gz" --cortex "C:\Users\33649\.brainstorm\tmp\brainsuite\bse_cortex_file.nii.gz"
1. cost(3,25,0.672,1)=72.583 
2. cost(3,25,0.704,1)=71.8706 
3. cost(3,25.625,0.688,1)=71.5842 
4. cost(3,25.625,0.704,1)=71.115 
1. cost(3,26.875,0.576,2)=74.0433 
2. cost(3,26.875,0.576,2)=74.0433 
3. cost(3,30,0.448,2)=54.2865 
4. cost(3,29.0625,0.384,2)=54.233 
lowest cost(3,29.0625,0.384,2)=54.233 
Wrote detailed brain mask C:\Users\33649\.brainstorm\tmp\brainsuite\bse_detailled_brain.mask.nii.gz 
Wrote cortex mask C:\Users\33649\.brainstorm\tmp\brainsuite\bse_cortex_file.nii.gz 
Wrote mask file C:\Users\33649\.brainstorm\tmp\brainsuite\bse_smooth_brain.mask.nii.gz 
Wrote skull-stripped MRI volume C:\Users\33649\.brainstorm\tmp\brainsuite\skull_stripped_mri.nii.gz 
BST> System call: bfc -i "C:\Users\33649\.brainstorm\tmp\brainsuite\skull_stripped_mri.nii.gz" -o "C:\Users\33649\.brainstorm\tmp\brainsuite\output_mri.bfc.nii.gz" -L 0.5 -U 1.5
Wrote corrected image C:\Users\33649\.brainstorm\tmp\brainsuite\output_mri.bfc.nii.gz 
BST> System call: bdp "C:\Users\33649\.brainstorm\tmp\brainsuite\output_mri.bfc.nii.gz" --tensor --nii "C:\Users\33649\Downloads\simnibs_examples\ernie\org\ernie_dMRI.nii" --t1-mask "C:\Users\33649\.brainstorm\tmp\brainsuite\bse_smooth_brain.mask.nii.gz" -g "C:\Users\33649\Downloads\simnibs_examples\ernie\org\ernie_dMRI.bvec" -b "C:\Users\33649\Downloads\simnibs_examples\ernie\org\ernie_dMRI.bval"
 
BDP Version: 19b (build #0077), released 2019-12-31 
 
================================================================================ 
                         Setting up dataset and inputs 
================================================================================ 
Reading input flags... 
 
Checking input files... 
 
Processing data with fileprefix: 
	C:\Users\33649\.brainstorm\tmp\brainsuite\output_mri 
 
================================================================================ 
                   Co-registration and Distortion Correction 
================================================================================ 
Reading the input parameters for co-registration... 
 
Total usable memory found:   6.28GB 
 
Checking orientation information...Done 
Extracting 0-diffusion (b=0) image from input DWIs... 
 
DWI mask is not defined in input flags. BDP will try to estimate a (pseudo) mask 
from 0-diffusion (b=0) image. Automatic mask estimation may not be accurate in 
some sitations and can affect overall quality of co-registration. In case 
co-registration is not accurate, you can define a DWI mask by using flag 
--dwi-mask <mask_filename>. The mask can be generated and hand edited in 
BrainSuite interface. This mask would be used only for registration purposes 
(and not for statistics computation). 
Saved (pseudo) masks: C:\\Users\\33649\\.brainstorm\\tmp\\brainsuite\\output_mri.dwi.RAS.mask.nii.gz 
 
 
Starting Registration based distortion Correction... 
Loading data... 
 
Starting rigid registration of input images... 
Reading input data... 
Setting/generating masks... 
Matching centroids (approx. align)... 
Matching resolution of the images... 
Normalizing intensity of images...  
Performing affine registration with 6 degrees of freedom... 
Search-based initialization...  
[... 932 lines cut ...]
Rigid registration done. 
 
Setting up data for non-rigid registration based distortion correction... 
Correcting intensity non-uniformity in b=0 image...Index exceeds the number of array elements (0). 
 
Error in find_bounding_box (line 16) 
 
Error in EPI_correct_files_registration_INVERSION>trim_msk_PED (line 513) 
 
Error in EPI_correct_files_registration_INVERSION (line 271) 
 
Error in coregister_diffusion_mprage_pipeline (line 219) 
 
Error in BrainSuite_Diffusion_pipeline (line 26) 
 
MATLAB:badsubscript 

***************************************************************************
** Error: Cannot convert files: 
** C:\Users\33649\Downloads\simnibs_examples\ernie\org\ernie_dMRI.nii
** C:\Users\33649\Downloads\simnibs_examples\ernie\org\ernie_dMRI.bval
** C:\Users\33649\Downloads\simnibs_examples\ernie\org\ernie_dMRI.bvec
** BrainSuite failed at step 3/3 (BDP).
** Check the Matlab command window for more information.
***************************************************************************

ajoshiusc, 2020-08-20:
BDP is not able to find a bounding box for the images. My guess is that there is severe artefact or the data is blank for some directions.

tmedani, 2020-09-03:
Hi,
just an update, 
this is error is solved by @ajoshiusc 
this error may appear in very specific cases (limited cases), where the MRI has big FOV where the neck and its bottom is included. 

@ajoshiusc will document how to solve this problem on the BrainSuite web site and we will add the link to the Brainstorm tutorial in the troubleshooting section. 
https://neuroimage.usc.edu/brainstorm/FemTensors?highlight=%28Troubleshooting%29

tmedani, 2020-10-14:
@ajoshiusc were you able to document this issue and how to solve it with Brainsuite?
we will add the link to the documentation here and on the Brainstorm tutorial https://neuroimage.usc.edu/brainstorm/FemTensors#Troubleshooting

ftadel, 2021-03-13:
@tmedani Is this issue still up-to-date?

tmedani, 2021-03-13:
yes, @ajoshiusc should write a report about this error and put it in the Brainsuite website, and then we will refer to it if this issue appears, it's mainly due to the neck.

let's wait for Anand input,

ftadel, 2021-09-19:
Any update with this issue?

tmedani, 2021-09-22:
This is a specific case that appeared only with Ernie subject.

closed 2021-09-22 by tmedani