brainstorm-tools/brainstorm3 #327: Problem with SimNIbs/headreco when the MRI is cuted

tmedani, 2020-08-26:
Hi @ftadel, 

I have noticed a new problem with headreco after the new modification that cut the MRI.

Since headreco uses the probabilistic template, the cut part of the input MRI is replaced by this template, 
here is the result when we use the default cut value,

[image]

I have discussed this error with SimNibs team and the solution is, instead to cut the MRI, we need to reduce the field of view (FOV) of the MRI to only the top part of the head that we need.
https://github.com/simnibs/simnibs/issues/30 

I don't have the Matlab code to do it right now, and from my understanding, this will affect the coordinate system and it needs to introduce new corrections in the CS that compensate the removed part.  

I don't know how much work it needs and when I can do it.... and if it's worth it

In meantime, we may deactivate the zneck for the SimNibs method.

ftadel, 2020-08-30:
Reducing the volume FOV without reslicing (= simply cutting slices along the Z axis) is not a solution equivalent to the cutting the neck with a plane defined in MNI space. It depends on the orientation of the head...

I can imagine 2 possibilities:

### Solution 1

While creating the .nii volume that is passed to SIMNIBS, instead of setting to zero the voxels, we could cut the Z slices that are below the lowest head voxel (= voxel with an intensity > detected background level) left after cutting with the current Zneck method. This would take the format of an additional parameter in the function process_mri_deface/CutMriPlane.

Is there a huge computation time in SIMNIBS:
1) with and without the neck cut by Brainstorm? (Zneck=-115 vs. Zneck=0)
2) with and without the lowest Z-slices?  (you can directly edit the MRI volume structure in the Brainstorm subject: `sMRI.Cube(:,:,iLowZ:end) = [];`)

### Solution 2

After calling SIMNIBS, we simply cut the neck from the FEM mesh at the same Zneck level (we already have the code for that: fem_resect.m).
Can you check whether this gives a good result? => just right-click on the FEM mesh you posted > Resect neck
If it is satisfying, we just add a call to fem_resect after calling SIMNIBS in process_fem_mesh.

tmedani, 2020-08-31:
> Reducing the volume FOV without reslicing (= simply cutting slices along the Z axis) is not a solution equivalent to the cutting the neck with a plane defined in MNI space. It depends on the orientation of the head...

I think just cutting along the Z ==> set to zero the voxel bellow the Z plan will not reduce the FOV.
The image will still have the same size with zeros at the 'removed' voxels. 
Within Headreco this FOV will be filled by the probabilistic template. 

I have tried with 
 sMri.Cube(maskCut) = [];
https://github.com/brainstorm-tools/brainstorm3/blob/ca16e9676efaa200ba95cfb9138142ab46341282/toolbox/process/functions/process_mri_deface.m#L215

But the output is not in good format,  I will try with sMRI.Cube(:,:,iLowZ:end) = [];  and will let you know.

> ### Solution 1
> While creating the .nii volume that is passed to SIMNIBS, instead of setting to zero the voxels, we could cut the Z slices that are below the lowest head voxel (= voxel with an [...]

I think this is a good option to investigate, however, reducing the bottom part of the FOV requires modification on the coordinate systems, right?

> Is there a huge computation time in SIMNIBS:
> 
> 1. with and without the neck cut by Brainstorm? (Zneck=-115 vs. Zneck=0)

I did not check, but I guess yes, since there is a huge part of the volume which is not included in the segmentation and the mesh, I think the meshing part takes more time. 

> 2. with and without the lowest Z-slices?  (you can directly edit the MRI volume structure in the Brainstorm subject: `sMRI.Cube(:,:,iLowZ:end) = [];`)

I will check it and share with you the results. 

> ### Solution 2
> After calling SIMNIBS, we simply cut the neck from the FEM mesh at the same Zneck level (we already have the code for that: fem_resect.m).
> Can you check whether this gives a good result? => just right-click on the FEM mesh you posted > Resect neck

we have already checked this part and it works fine. 

> If it is satisfying, we just add a call to fem_resect after calling SIMNIBS in process_fem_mesh.

I need to run the FEM computation just for validation of the results.

ftadel, 2020-09-01:
> I think this is a good option to investigate, however, reducing the bottom part of the FOV requires modification on the coordinate systems, right?

Indeed, it's not straightforward... it requires to edit the volume transformation matrices.

> > with and without the neck cut by Brainstorm? (Zneck=-115 vs. Zneck=0)
> I did not check, but I guess yes, since there is a huge part of the volume which is not included in the segmentation and the mesh, I think the meshing part takes more time.

Do you know at which stage the missing part of the head is added?
Does it add the missing part **of the volume** and then mesh it, rather than meshing the volume in input and then adding some template mesh for the missing parts of the head?

> > If it is satisfying, we just add a call to fem_resect after calling SIMNIBS in process_fem_mesh.
> I need to run the FEM computation just for validation of the results.

It would be good to have an evaluation of the difference of SIMNIBS computation time between solutions #1 and #2 
If #1 is much faster we'll go for this one, if it's not, we'll go for #2.

ftadel, 2021-03-13:
@tmedani Is this issue still up-to-date?

tmedani, 2021-03-13:
long story, but the take-home message  is to not cut the MRI before if we want to use the headreco,
 and we will cut the mesh after this process. 

we have really built a super FEM tool in Brainstorm, I like it. 

I have still some features in mind, and in my local version, that I may push one day.

closed 2021-03-13 by tmedani