brainstorm-tools/brainstorm3 #328: CS convert for headreco output oversizes the mesh in some direction

tmedani, 2020-08-26:
Hello,

New strange behavior of the cs_convert in this specific case with the  SimNibs mesh output, 

Here is the case, I'm using the MRI from the BrainSuite tutorial, run the SimNibs for mesh generation, and get the head model. 

Here is the result

[image]

[image]

The mesh is oversized in some direction compare to the input MRI, and to the tissue. 

After some investigations,  all the MRI and the tissues are correct and well-aligned
[image]

The original mesh output from SimNibs also looks correct, 

[image]

 
It seems that the error is occurred at the step of the coordinate system conversion, 

[image]

https://github.com/brainstorm-tools/brainstorm3/blob/f81bda70c9ebad55d5b5ba8d472f2c054936e5ea/toolbox/io/in_tess.m#L190

Here is the output of these lines, 

[image]

The problem is mainly at this line 
            TessMat.Vertices = cs_convert(sMri, 'voxel', 'mri', TessMat.Vertices);

I think the problem comes from the voxelsize, =    [1.33, 1, 1], 

I don't have a solution right now without changing the core function of the cs_convert. 

But, maybe a general solution is to reslice the initial MRI to an isotropic voxel and then call headreco?

In case, where someone wants to reproduce this issue, here is the link to the sMRI and the TessMat : https://drive.google.com/open?id=1mLJ3zfzFZFL5_K-XQGWPK-u_cdj0Tl6x&authuser=medani%40usc.edu&usp=drive_fs

ftadel, 2020-08-30:
It is not a problem with cs_convert, it is a problem with the import of the SIMNIBS outputs.
What it means is that the vertices of the mesh read from the .msh files are not in "voxel coordinates".
The line `TessMat.Vertices = cs_convert(sMri, 'voxel', 'mri', TessMat.Vertices);` multiplies the coordinates with sMRI.Voxelsize and divides by 1000. It has to be replaced with something else, maybe only a division by 1000.

You can probably easily fix this. Check that it works on a few MRI with different voxel sizes.

tmedani, 2020-08-31:
> The line `TessMat.Vertices = cs_convert(sMri, 'voxel', 'mri', TessMat.Vertices);` multiplies the coordinates with sMRI.Voxelsize and divides by 1000. It has to be replaced with something else, maybe [...]

I think so, it does not require the multiplication by the Voxelsize, 

I will check it, but it need time to validate it in some MRI.

ftadel, 2020-09-01:
Let me know if it works to replace the line
`TessMat.Vertices = cs_convert(sMri, 'voxel', 'mri', TessMat.Vertices);`
with:
`TessMat.Vertices = TessMat.Vertices ./ 1000;`

I will commit the change after you validate with a volume with a voxel size different from [1,1,1].

tmedani, 2020-09-02:
Hi @ftadel 
I check in some MRI (4 subjects)  with different voxel size, and the error comes from the 

cs_convert(sMri, 'voxel', 'mri', TessMat.Vertices)

please just replace it 
> with:
> `TessMat.Vertices = TessMat.Vertices ./ 1000;`

It solves the problem, 

However, I don't understand why the 0.5 voxel correction in line 194,
I do not notice any visual difference with and without it.
Is there any reason for adding it?  @juangpc

ftadel, 2020-09-02:
>please just replace it

Done: https://github.com/brainstorm-tools/brainstorm3/commit/a864cce53a4e7fa65688234c1decfc7c75b433a7

> However, I don't understand why the 0.5 voxel correction in line 194,

Probably a different in the convention of where the center of the first voxel is: (0,0,0) or (0.5,0.5,0.5) ?

tmedani, 2020-09-02:
> > please just replace it
> 
> Done: [a864cce](https://github.com/brainstorm-tools/brainstorm3/commit/a864cce53a4e7fa65688234c1decfc7c75b433a7)

super, thanks.

> > However, I don't understand why the 0.5 voxel correction in line 194,
> 
> Probably a different in the convention of where the center of the first voxel is: (0,0,0) or (0.5,0.5,0.5) ?

If I understand well, in this case, we move the whole head by (0.5/1000) mm ?
which seems to be insignificant.

closed 2020-09-02 by tmedani