brainstorm-tools/brainstorm3 #336: Include the Gain matrix in "Apply the montage" process

tmedani, 2020-10-01:
Hi @ftadel 

we (@juangpc & @jcmosher ) have noticed that when the process "Apply montage" is called, the Gain matrix is not included in the final results.

I did not check the code but we are wondering if it's possible to add it to this process.  

[image]

[image]

[image]

Let us know how we can help.

Thanks.

ftadel, 2020-10-01:
Something like this?
https://github.com/brainstorm-tools/brainstorm3/commit/ed913f86aceeb7ee0e829e2896b3245314310e7e#diff-e5c8e6bc456202d4a14872df359a1a9cR277

(please close the issue if you're satisfied)

tmedani, 2020-10-01:
wow!
I will check it asap, and let you know. 

Thanks

tmedani, 2020-10-14:
Hi @ftadel 

Sorry for my late answer, my local version is far from the master version, I will push soon the new options.  

I was able to test your changes and I'm not getting what is expected, the transformation is not applied for the head model.

[image]

tmedani, 2020-10-14:
Oops,

 It's working when I change the data and the montage from Longitudinal to bipolar,

[image]

Is it working only with the bipolar option?

tmedani, 2020-10-14:
Does it make sense to apply the montage also to the "Noise Covariance matrix"? 
and generate the associated new covariance matrix?

ftadel, 2020-10-14:
>Is it working only with the bipolar option?

It should work with all the headmodels, but only if the option "Create new folders" is selected and the subject is not using a channel file shared across multiple folders.
https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/process/functions/process_montage_apply.m#L278

If you observe any error, please try to give me a way to reproduce it with a simple example.

> Does it make sense to apply the montage also to the "Noise Covariance matrix"?

I thought about it, it would be very easy to implement (more or less a copy paste of the head model section).
But in the noise covariance matrix, there is not only the linear part (NoiseCov) but a non-linear measure (**FourthMoment**). Computing the noise covariance from the original recordings and then applying the montage would lead to different results than applying the montage and then computing the noise covariance file. Because of this I decided not to include it.

Maybe this is doable, but we need the decisions of an expert for this:
@jcmosher @dimitriospantazis @sbaillet ?

tmedani, 2020-10-14:
> It should work with all the headmodels, but only if the option "Create new folders" is selected and the subject is not using a channel file shared across multiple folders. [...]

Ok, I will check it, currently, I don't understand why this common file appears even when I have only one folder.
I will recheck the tutorial

Looking forward to the comments of our experts :)

dimitriospantazis, 2020-10-14:
Hi @tmedani,

Estimation of the covariance matrix is a non-linear operation, so it cannot commute with a linear change of the montage. So Francois is right to require recomputing it.

Cheers,
Dimitrios

tmedani, 2020-10-15:
Hi @dimitriospantazis 

Thanks for the confirmation, it always better to have a confirmation. 
 I will do it on my data. 

@ftadel from the users' side, 
is there a way to include this computation from the "new montage raw signal" within the montage process? 

otherwise, we can close this issue :)

Thanks to all of you,
Best,

jcmosher, 2020-10-15:
Agree. Covariance needs to be recalculated for the new montage.

tmedani, 2020-10-26:
> @tmedani
> 
> > I don't understand why this common file appears even when I have only one folder.
> 
> You probably configured your subject to a shared channel file.
> Right-click on the subject > Edit subject > **Default channel file : NO**

d'accord, got it, thank you. 

> > is there a way to include this computation from the "new montage raw signal" within the montage process?
> 
> I'm not sure I understand this correctly.
> Can you describe better your request?

I meant if it's possible to add the computation of the new covariance in the process from the recording

closed 2020-10-31 by ftadel

Moo-Marc, 2020-10-31:
> The recommended workflow is to apply the montages/re-referencing among the early stages of the pre-processing: after the frequency filters, but before the ICA/SSP

I'm a bit confused. I didn't follow the discussion, but I had in my to-do list to modify the "apply montage" process to allow raw inputs.  Here you seem to imply it's already doable?  Or you're referring to another function and not the process?

ftadel, 2020-11-01:
@Moo-Marc I'm sorry, I got confused... It is recommended to re-reference the EEG in the early preprocessing, but indeed, it is not possible to call the process "Apply montage" on continuous files. This could be a useful feature though.

At the moment, the process "Apply montage" must be called after importing the epochs to the database.
But I don't think we have any use case where we would recommend to call it after estimating the sources (forward model, noise covariance, inverse model).