brainstorm-tools/brainstorm3 #339: update fails

christian-oreilly, 2020-10-07:
On starting brainstorm, I get: 
```BST> Checking internet connectivity... failed```
This is due to `url_read_fcn('http://neuroimage.usc.edu/bst/getversion.php')` generating an exception, causing `bst_check_internet()` to return at line 37. Since for me `bst_get('MatlabVersion') == 908`, `url_read_fcn` uses `webread` which fails with 

```
>> url_read_fcn('http://neuroimage.usc.edu/bst/getversion.php')
Error using webread (line 119)
Expected input to be one of these types:

char

Instead its type was string.
```

In case it might be relevant, this issue was found on a Ubuntu distribution (18.04).

ftadel, 2020-10-07:
What do you get if you type:
`webread('http://neuroimage.usc.edu/bst/getversion.php')`
and:
`urlread('http://neuroimage.usc.edu/bst/getversion.php')`
?

christian-oreilly, 2020-10-07:
The `webread` one I gave in the previous post, the `urlread` one works fine for me.

```
>> urlread('http://neuroimage.usc.edu/bst/getversion.php')

ans =

    '% Brainstorm
     <BR>% v. 3.201007 (07-Oct-2020)<BR>'

```

Maybe using a monkey-patch approach that would make the code rely less on versions:

```
try
    str = webread('http://neuroimage.usc.edu/bst/getversion.php');
catch
    str = urlread('http://neuroimage.usc.edu/bst/getversion.php');
end
```

christian-oreilly, 2020-10-07:
With this change (and with a change in permissions of /usr/local where the script tries to save the zip file), the update worked on my setup.

ftadel, 2020-10-12:
The problem is not in the function webread, but in the input parameters that are given to it...
Maybe there is something to fix in the way the function webread is called.
Since urlread is being deprecated, it would be better to find a way to fix this correctly to keep using webread in recent versions of Matlab.

Unfortunately, I can't reproduce this error, so I can't debug it myself...
If you put a breakpoint at line 119 in webread.m, can you manage to track the cause of the problem?
Why is there a variable that is "string" instead of "char"?

@martcous do you have a Ubuntu 18 + Matlab 2020a somewhere?

ftadel, 2020-10-12:
@Moo-Marc same question: do you have a Ubuntu 18 + Matlab 2020a somewhere?

Moo-Marc, 2020-10-13:
We're still on 2019b, but I'll ask for it.

martcous, 2020-10-16:
We just got 2020b installed and both functions work fine on my end (Ubuntu 18.04 + Matlab 2019b or 2020b). I'll try to install 2020a on my workstation to try to reproduce the issue.

martcous, 2020-10-16:
It was easier than expected to install, I should have tested this sooner sorry. I cannot reproduce the issue on Ubuntu 18.04 + Matlab 2020a either, both webread() and urlread() functions work fine on my end. I think the try.. catch solution with the non-deprecated function attempted first (webread) is a passable solution.

ftadel, 2020-10-22:
What about this fix? 
https://github.com/brainstorm-tools/brainstorm3/commit/ef875f54f629303fa14d29e3da009575ed695c70
It always tests both webread and urlread, but with a different order depending on the Matlab version.

@christian-oreilly 
Please close this issue if you think it is resolved.

ftadel, 2020-11-09:
@christian-oreilly  Is this working now?

christian-oreilly, 2020-11-09:
Looks good to me! Thanks @ftadel

closed 2020-11-09 by christian-oreilly