brainstorm-tools/brainstorm3 #345: Run Multiple Instances of Brainstorm (from multiple instances of Matlab)

tmedani, 2020-10-19:
Hi, 

I'm just wondering if it is possible to run different Brainstorm instances (from different Matlab sessions) without "overlapping". 

Is it sufficient to change just the temporary folder (specific folder for each instance)?

or this is not recommended at all?

@ftadel @martcous 
Thanks

juangpc, 2020-10-19:
Nice question. 
I'd like to guesstimate: I think that the brainstorm figures all rely on a singleton pattern right? I mean all the BstFrame context. For instance how would bst_progress know which bst is calling him? And the use of the dbfolder doesn't mutex for several clients at the same time. Right? Also the is_started.txt file? 
So if I were to bet, I'd bet that multiple instances would not work properly. 
But, I'll let @ftadel @martcous or whoever prove me wrong (again 😃 ).

tmedani, 2020-10-19:
Thanks, Juan, 

Right now I'm using two instances from two different versions of Brainstorm on two different protocols, 
and the basic functionalities are working (mainly the displays), I didn't perform any process that involves the temp folder or change on the DB.
  
[image]

juangpc, 2020-10-19:
:hear_no_evil:

martcous, 2020-10-19:
Hi both!

Is this officially supported or recommended? No, although we're working on this.

Is it possible? Yes.. at your own risk. I do it all the time actually to quickly test things between different versions of Matlab or Brainstorm. The singleton pattern is not a problem since the memory of both instances is not shared as they are inside separate instances of Matlab itself. However, as you know, it's not supported to work on the same protocol from separate instances: the database would need to be refreshed manually at every operation and Brainstorm cannot guarantee that the same file won't be modified by 2 or more processes at the same time, potentially causing issues and corrupting the database. As you know, I'm actively working on this at the moment.

Working on different protocols should be OK, just keep in mind that your instances of Brainstorm will share the .brainstorm folder where preferences and the last state of the software is saved, so the state and preferences of Brainstorm stored will be from the last instance of Brainstorm you closed (since they are overwritten at each shut down of the software).

To clarify: this is NOT recommended and you should keep in mind there is a risk of corruption or otherwise unexpected behavior on your protocols, but possible for quick testing and debugging.

I'll let François chime in and correct me if I'm wrong.

Cheers!

mpompolas, 2020-10-20:
I've been using this for computationally intense tasks (e.g. TF decompositions) - I usually open the same number of instances as the number of cores on my computer.
The approach I follow for assigning each task on each instance is that they work on separate folders (different subject or different trial-tag etc) so you avoid file modifications as @martcous described above.

tmedani, 2020-10-21:
Hi,

thanks, @martcous & @mpompolas for your feedback. 

I'm using these different versions as @martcous, in order to check the code and the results before PR or a Pull, But I have never tried to perform any computation on the DB,  since it's risky. 
 
 
@mpompolas 
> The approach I follow for assigning each task on each instance is that they work on separate folders (different subject or different trial-tag etc) so you avoid file modifications as @martcous described above.
> Never tried multiple versions that @tmedani has on the screenshot though

Interesting, then it's possible to run some computations that use the "local memory" of the Matlab instance,
but not possible if it uses the intermediate data within the .brainstom/tmp folder.

ftadel, 2020-10-22:
> Interesting, then it's possible to run some computations that use the "local memory" of the Matlab instance,
> but not possible if it uses the intermediate data within the .brainstom/tmp folder.

If you respect the following conditions, you can run multiple instances on the same computer from the same user:
1) you are not processing the same subjects at the same time
2) you always reload the entire protocol when you start a new instance of Brainstorm
3) your processes do not rely on anything saved in .brainstorm/tmp (it excludes most of the processes from external programs, like SPM, BrainSuite, OpenMEEG, SimNIBS, CAT...)

Executing processes that store files in the .brainstorm/tmp folder is possible by running Brainstorm from different users on the same computer.

closed 2020-10-27 by tmedani