brainstorm-tools/brainstorm3 #352: Bugs integration CAT12 v1728

ftadel, 2020-11-01:
### 1. catROIs_CAT12_...xml vs catROI_CAT12_...xml

Execution of the following segmentation batch:
```
    matlabbatch{1}.spm.tools.cat.estwrite.data = {[NiiFile ',1']};
    matlabbatch{1}.spm.tools.cat.estwrite.data_wmh = {''};    % NEW 1728 
    matlabbatch{1}.spm.tools.cat.estwrite.useprior = '';      % NEW 1728
    matlabbatch{1}.spm.tools.cat.estwrite.nproc = 0;   % Blocking call to CAT12
    matlabbatch{1}.spm.tools.cat.estwrite.opts.tpm = {TpmNii};
    matlabbatch{1}.spm.tools.cat.estwrite.opts.affreg = 'mni';
    matlabbatch{1}.spm.tools.cat.estwrite.opts.biasstr = 0.5;   % OLD
    matlabbatch{1}.spm.tools.cat.estwrite.opts.biasacc = 0.5;   % NEW 1716
    matlabbatch{1}.spm.tools.cat.estwrite.extopts.APP = 1070;
    matlabbatch{1}.spm.tools.cat.estwrite.extopts.spm_kamap = 0;   % NEW 1716
    matlabbatch{1}.spm.tools.cat.estwrite.extopts.LASstr = 0.5;
    matlabbatch{1}.spm.tools.cat.estwrite.extopts.gcutstr = 2;
    matlabbatch{1}.spm.tools.cat.estwrite.extopts.WMHC = 1;   % NEW 1716
    matlabbatch{1}.spm.tools.cat.estwrite.extopts.registration.dartel.darteltpm = {dartelTpm};   % OLD
    matlabbatch{1}.spm.tools.cat.estwrite.extopts.registration.shooting.shootingtpm = {shootingTpm};  % NEW 1716 
    matlabbatch{1}.spm.tools.cat.estwrite.extopts.registration.shooting.regstr = 0.5;  % NEW 1716 
    matlabbatch{1}.spm.tools.cat.estwrite.extopts.vox = 1.5;
    matlabbatch{1}.spm.tools.cat.estwrite.extopts.restypes.optimal = [1 0.1];  % NEW 1716 
    matlabbatch{1}.spm.tools.cat.estwrite.extopts.ignoreErrors = 1;  % NEW 1716 
    matlabbatch{1}.spm.tools.cat.estwrite.output.GM.native = 1;
    matlabbatch{1}.spm.tools.cat.estwrite.output.GM.mod = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.GM.dartel = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.WM.native = 1;
    matlabbatch{1}.spm.tools.cat.estwrite.output.WM.mod = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.WM.dartel = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.TPMC.native = 1;
    matlabbatch{1}.spm.tools.cat.estwrite.output.TPMC.warped = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.TPMC.mod = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.TPMC.dartel = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.CSF.native = 1;
    matlabbatch{1}.spm.tools.cat.estwrite.output.CSF.warped = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.CSF.mod = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.CSF.dartel = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.label.native = 1;
    matlabbatch{1}.spm.tools.cat.estwrite.output.label.warped = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.label.dartel = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.labelnative = 1;
    matlabbatch{1}.spm.tools.cat.estwrite.output.bias.warped = 1;
    matlabbatch{1}.spm.tools.cat.estwrite.output.jacobianwarped = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.warps = [1 1];      % Changed from [0 0] with NEW 1728
    matlabbatch{1}.spm.tools.cat.estwrite.output.rmat = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.surface = 1;
    matlabbatch{1}.spm.tools.cat.estwrite.output.surf_measures = 1;
    matlabbatch{1}.spm.tools.cat.estwrite.output.ROImenu.atlases.neuromorphometrics = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.ROImenu.atlases.lpba40 = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.ROImenu.atlases.cobra = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.ROImenu.atlases.hammers = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.ROImenu.atlases.ownatlas = {''};
```

Runs correctly, but prints an error message:
```
  Remove "C:\Users\franc\.brainstorm\tmp\cat12\label\catROI_CAT12_1001_2020-11.xml" from dependency list because it does not exist!
One or multiple files do not exist and were removed from the dependency list 
and following batches will may not work correctly!
```

The file `C:\Users\franc\.brainstorm\tmp\cat12\label\**catROIs**_CAT12_1001_2020-11.xml`  exists.
But not `C:\Users\franc\.brainstorm\tmp\cat12\label\**catROI**_CAT12_1001_2020-11.xml`
Is there an error somewhere?

ftadel, 2020-11-01:
@i5gach FYI

ftadel, 2020-11-10:
Updated the script that runs CAT12 in Brainstorm in this commit:
https://github.com/brainstorm-tools/brainstorm3/commit/5212b573192212cc661caf9661b459323ae7f469

The SPM batch generated by Brainstorm is now the following:
```
    matlabbatch{1}.spm.tools.cat.estwrite.data = {'C:\Users\franc\.brainstorm\tmp\cat12\CAT12_ICBM152.nii,1'};
    matlabbatch{1}.spm.tools.cat.estwrite.data_wmh = {''};
    matlabbatch{1}.spm.tools.cat.estwrite.useprior = '';
    matlabbatch{1}.spm.tools.cat.estwrite.nproc = 0;                % Blocking call to CAT12
    matlabbatch{1}.spm.tools.cat.estwrite.opts.tpm = {'C:\Users\franc\.brainstorm\defaults\spm\TPM.nii'};      % User-defined TPM atlas
    matlabbatch{1}.spm.tools.cat.estwrite.output.warps = [1 1];     % Save deformation fields: [forward inverse]
    matlabbatch{1}.spm.tools.cat.estwrite.output.GM.native   = 1;   % GM tissue maps
    matlabbatch{1}.spm.tools.cat.estwrite.output.GM.warped   = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.GM.mod      = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.GM.dartel   = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.WM.native   = 1;   % WM tissue maps
    matlabbatch{1}.spm.tools.cat.estwrite.output.WM.warped   = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.WM.mod      = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.WM.dartel   = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.TPMC.native = 1;   % Tissue classes 4-6 to create own TPMs
    matlabbatch{1}.spm.tools.cat.estwrite.output.TPMC.warped = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.TPMC.mod    = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.TPMC.dartel = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.CSF.native  = 1;   % CSF tissue maps
    matlabbatch{1}.spm.tools.cat.estwrite.output.CSF.warped  = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.CSF.mod     = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.CSF.dartel  = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.label.native = 1;  % Label: background=0, CSF=1, GM=2, WM=3, WMH=4
    matlabbatch{1}.spm.tools.cat.estwrite.output.label.warped = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.label.dartel = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.ROImenu.atlases.neuromorphometrics = 0;  % No volume atlases
    matlabbatch{1}.spm.tools.cat.estwrite.output.ROImenu.atlases.lpba40             = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.ROImenu.atlases.cobra              = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.ROImenu.atlases.hammers            = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.surface = 2;   % 1: lh+rh, 2:lh+rh+cerebellum
    matlabbatch{1}.spm.tools.cat.estwrite.output.surf_measures = 1;   % Cortical thickness

    matlabbatch{2}.spm.tools.cat.stools.surfextract.data_surf(1) = cfg_dep('CAT12: Segmentation (current release): Left Central Surface', substruct('.','val', '{}',{1}, '.','val', '{}',{1}, '.','val', '{}',{1}, '.','val', '{}',{1}), substruct('()',{1}, '.','lhcentral', '()',{':'}));
    matlabbatch{2}.spm.tools.cat.stools.surfextract.GI = 1;     % Gyrification index
    matlabbatch{2}.spm.tools.cat.stools.surfextract.SD = 1;     % Sulcal depth
    matlabbatch{2}.spm.tools.cat.stools.surfextract.nproc = 0;  % Blocking call to CAT12
```

However, all the errors mentioned above are still there.
Additionally, I confirm that when running the atlas projection with the GUI menu **Surface Tools > Map Surface Atlas to Individual Surface**, we get exactly the same errors as above:
```
Error using cat_check_system_output (line 27)

Error in cat_surf_map_atlas (line 47)
      [ST, RS] = cat_system(cmd); cat_check_system_output(ST,RS);

Error in spm (line 1004)
    evalin('base',CB)
 
Error while evaluating UIControl Callback.
```

ftadel, 2020-11-11:
Additional execution example on the MNI ICBM152 template:
https://neuroimage.usc.edu/forums/t/cat12-missing-files/23121/3

i5gach, 2020-11-15:
@ftadel 

### 1. catROIs_...xml vs catROI_...xml ?

In order to disable ROI processing for volumes the ROI structure should have the following syntax:
`matlabbatch{1}.spm.tools.cat.estwrite.output.ROImenu.noROI = struct([]);`

Please note that surface extraction for the cerebellum is still experimental and does not reliably work and should only be used with great care. Currently, I cannot recommend regular use and this should be disabled which also holds for the experimental surface_measure flag (thickness will be saved anyway):
`matlabbatch{1}.spm.tools.cat.estwrite.output.surface = 1;`
`%matlabbatch{1}.spm.tools.cat.estwrite.output.surf_measures = 1;`

### 2. CAT_ResampleSurf: Vertex index out of range
I couldn't get this to work on Windows systems, and after trying for days to solve the problem, I gave up and implemented an alternative approach. This is available in r1732.

ftadel, 2020-11-16:
> - In order to disable ROI processing for volumes the ROI structure should have the following syntax:
> - Please note that surface extraction for the cerebellum is still experimental
> - also holds for the experimental surface_measure flag (thickness will be saved anyway)

Thank you for your comments.
I made the cerebellum extraction an discreet option (disabled by default) and adjusted the SPM batch based on your recommendations. We might need the deformation fields and the volume atlases in the future, but not immediately.
https://github.com/brainstorm-tools/brainstorm3/commit/0f6ad98d6d613f36d8fcc4cb3aa2baaf88572593

The "advanced" interface in Brainstorm now looks like this:
[image]

And the default batch is the following:
```
    matlabbatch{1}.spm.tools.cat.estwrite.data = {'C:\Users\franc\.brainstorm\tmp\cat12\CAT12_1001_2020_11_1732.nii,1'};
    matlabbatch{1}.spm.tools.cat.estwrite.data_wmh = {''};
    matlabbatch{1}.spm.tools.cat.estwrite.useprior = '';
    matlabbatch{1}.spm.tools.cat.estwrite.nproc = 0;                % Blocking call to CAT12
    matlabbatch{1}.spm.tools.cat.estwrite.opts.tpm = {'C:\Users\franc\.brainstorm\defaults\spm\TPM.nii'};      % User-defined TPM atlas
    matlabbatch{1}.spm.tools.cat.estwrite.output.bias.warped = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.warps       = [0 0]; % Skip deformation fields for the moment: [forward inverse]
    matlabbatch{1}.spm.tools.cat.estwrite.output.GM.native   = 1;   % GM tissue maps
    matlabbatch{1}.spm.tools.cat.estwrite.output.GM.warped   = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.GM.mod      = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.GM.dartel   = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.WM.native   = 1;   % WM tissue maps
    matlabbatch{1}.spm.tools.cat.estwrite.output.WM.warped   = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.WM.mod      = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.WM.dartel   = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.TPMC.native = 1;   % Tissue classes 4-6 to create own TPMs
    matlabbatch{1}.spm.tools.cat.estwrite.output.TPMC.warped = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.TPMC.mod    = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.TPMC.dartel = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.CSF.native  = 1;   % CSF tissue maps
    matlabbatch{1}.spm.tools.cat.estwrite.output.CSF.warped  = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.CSF.mod     = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.CSF.dartel  = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.label.native = 1;  % Label: background=0, CSF=1, GM=2, WM=3, WMH=4
    matlabbatch{1}.spm.tools.cat.estwrite.output.label.warped = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.label.dartel = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.ROImenu.noROI = struct([]);   % CGaser comment: Correct syntax to disable ROI processing for volumes   
    matlabbatch{1}.spm.tools.cat.estwrite.output.ROImenu.atlases.neuromorphometrics = 0;  % No volume atlases
    matlabbatch{1}.spm.tools.cat.estwrite.output.ROImenu.atlases.lpba40             = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.ROImenu.atlases.cobra              = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.ROImenu.atlases.hammers            = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.surface = 1;
```

> I couldn't get this to work on Windows systems, and after trying for days to solve the problem, I gave up and implemented an alternative approach. This is available in r1732.

Not sure how to make this work. There are issues with the handling of the file separators on Windows.
```
>> cat_surf_map_atlas('C:\Work\RawData\Test\fem\A1001_cat12_2020-11_1728\surf\lh.central.CAT12_1001_2020-11.gii', 'C:\Work\Dev\Divers\spm12\toolbox\cat12\atlases_surfaces\lh.aparc_a2009s.freesurfer.annot')
Error using cat_check_system_output (line 27)
Error:  could not open file "C:WorkRawDataTestfemA1001_cat12_2020-11_1728surflh.aparc_a2009s.txt".
System message: No such file or directory
Cannot read values in C:\Work\RawData\Test\fem\A1001_cat12_2020-11_1728\surf\lh.aparc_a2009s.txt.

Error in cat_surf_map_atlas (line 54)
      [ST, RS] = cat_system(cmd); cat_check_system_output(ST,RS);
```

This calls translates to the system call:
```
CAT_ResampleSurf -label 
"C:\Work\Dev\Divers\spm12\toolbox\cat12\templates_surfaces\lh.central.freesurfer.gii" 
"C:\Work\Dev\Divers\spm12\toolbox\cat12\templates_surfaces\lh.sphere.freesurfer.gii" 
"C:\Work\RawData\Test\fem\A1001_cat12_2020-11_1728\surf\lh.sphere.reg.CAT12_1001_2020-11.gii" 
"NULL" 
"C:\Work\RawData\Test\fem\A1001_cat12_2020-11_1728\surf\lh.aparc_a2009s.txt" "C:\Work\RawData\Test\fem\A1001_cat12_2020-11_1728\surf\lh.aparc_a2009s.CAT12_1001_2020-11.txt"
```

Note that there were issues on **Linux** as well: **not compatible with FreeSurfer's read_annotation.m**.
(I haven't checked if this was working with r1732).

> I rather use the surf folder for the annot files because this is necessary to display these files with the surf display tool in CAT12. It’s mandatory that texture (e.g. thickness) and annotation files are in the same folder for the surface files (which are necessary for displaying).

But then the output is not compatible with freesurfer processing tools. Other people using CAT outputs for other purposes than displaying them in CAT would have to move the files back to the `label` folder (they are actually labels and not surfaces).
I also noticed that you switched from the .annot to a .txt format.

Could you make these two changes optional?  
=> Additional input parameters to `cat_surf_map_atlas` to select between the **FreeSurfer-style (label/.annot)** and the **CAT-style (surf/.txt)**   (one single option for both or two separate options in )

i5gach, 2020-11-16:
Sorry for the mess. The new r1733 should hopefully run. 

The new settings look fine. These line scan be deleted because the ROI structure is set to not saving any ROI atlases:
    matlabbatch{1}.spm.tools.cat.estwrite.output.ROImenu.atlases.neuromorphometrics = 0;  % No volume atlases
    matlabbatch{1}.spm.tools.cat.estwrite.output.ROImenu.atlases.lpba40             = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.ROImenu.atlases.cobra               = 0;
    matlabbatch{1}.spm.tools.cat.estwrite.output.ROImenu.atlases.hammers            = 0;

ftadel, 2020-11-17:
Everything seems to be working both on Windows and Linux with r1733!
Only 45min for the segmentation+interpolation of the FsAverage atlases, this is unbeatable! :-)

I pushed the last modifications to the CAT12 wrapper + importer, it's all online:
https://github.com/brainstorm-tools/brainstorm3/commit/6de6b3d8d9e219d5e4bbd47f2fb029821781dc4d

Thank you very much for your help.

ftadel, 2020-11-17:
PS: I also changed the import code so that it looks for .annot file anywhere with the CAT segmentation folder, so it doesn't matter anymore if the .annot are in /surf or /label.

closed 2020-11-17 by ftadel