brainstorm-tools/brainstorm3 #353: Reading stimulus events and condition names from .nirs data

kdarti, 2020-11-02:
Hi, 

I have an enhancement request for the reading in of .nirs (Homer2 format) files. The .nirs file format allows for stimulus events in the **s** variable  and the names of experimental conditions in the **CondNames** variable (p.5, p.28. [Homer2 user's guide](http://www.nmr.mgh.harvard.edu/martinos/software/homer/HOMER2_UsersGuide_121129.pdf)). 

Currently [in_fopen_nirs_brs.m](/brainstorm-tools/brainstorm3/blob/master/toolbox/io/in_fopen_nirs_brs.m) does not read either either one of these variables from .nirs files. It would be great to have this feature implemented so that it's possible to read events and condition names stored in a .nirs file directly into nirstorm.

Edouard2laire, 2020-11-02:
Hi. 

PR #354 should work.   However, in the files you sent me, it seems to be a little difference between the event timing in the nirs file and the tsv file.  First tapping in the nirs file is at 111.4s (sample 1114) but is 111.3s in the tsv file. I don't know if that's important. 

Best regards, 
Edouard

Edouard2laire, 2020-11-02:
Hi.

PR #354 should work. 

Best regards,
Edouard

kdarti, 2020-11-02:
Many thanks Edouard!
 Was just about to reply about the time diff, but I see you fixed it now.

PR #354 (2020-11-02): NIRS - also read events from .nirs file
fix #353 

edit: I forgot Matlab array start at 1 and not 0; so we need to remove 1 to get the event starting time
closed 2020-11-02 by ftadel