brainstorm-tools/brainstorm3 #361: FIF file limits channel names to 15 Characters

bloyl, 2020-11-18:
FIF files currently limit channel names to 15 Characters which can restrict its use for some applications, particularly clinical eeg.

There is an MNE-python PR which solves this issue by shortening channel names and storing a renaming map so that during reading the shorten names are expanded to their original channel names.
See https://github.com/mne-tools/mne-python/issues/8312#issuecomment-700249945 and https://github.com/mne-tools/mne-python/pull/8348

We plan for a similar PR for MNE-matlab but before we move much farther we'd like to ensure that these files won't cause any issues with brainstorm or fieldtrip.

So I have 2 questions:
1) would the proposed change cause any fundamental problems for fif support within brainstorm?

2) would a PR to MNE-matlab be sufficient to support these files in brainstorm or are changes also needed within brainstorm code?

CC @larsoner @ftadel

ftadel, 2020-11-19:
> would the proposed change cause any fundamental problems for fif support within brainstorm?
> would a PR to MNE-matlab be sufficient to support these files in brainstorm or are changes also needed within brainstorm code?
No

bloyl, 2020-11-19:
Thanks. I'll let you know once there is an MNE-matlab PR to test against.

ftadel, 2021-03-13:
Done: https://github.com/brainstorm-tools/brainstorm3/commit/cf76e9feeb605f1a0d4838afe24d7383d8d9c618

closed 2021-03-13 by ftadel