brainstorm-tools/brainstorm3 #363: Bug icohere2019/lcohere2019: Bad evaluation of pValues

ftadel, 2020-11-19:
Following this forum post: https://neuroimage.usc.edu/forums/t/difference-between-averaging-coherence-and-concatenate/22726
I wanted to tell the users to use icohere2019 instead of icohere in their tests, but when testing it myself, I realized this function does not return any value most of the time.

Example with 30s of continuous MEG recordings from the introduction tutorial dataset, one pair of nearby sensors on the left (MLT16, MLT23) and the symmetrical sensors on the right (MRT16, MRT23).

Options selected:
[image]

Output for the different methods, in this order: **mscohere**, **icohere** (old), **icohere2019**, **lcohere2019**:
[image]
[image]
[image]
[image]

When looking at the pValues evaluated for icohere2019 and lcohere2019, we get almost all values > 0.9.
[image]

@HosseinShahabi Can you please double-check this is really the expected result?

HosseinShahabi, 2020-11-19:
There are a couple of issues here: 

for the new measures, there is no computation of the p-value, so what you see here is not meaningful. (I thought I made it clear previously but If not that's my fault)
 
Since we have other statistical tests later in the process and planned to unify all the tests on the connectivity matrix elements in a separate step, I decided not to add something extra here. I'm not sure the previous one is practical either.

I see here that lagged and imaginary 2019 both have the exact same response? That cannot be true at least for all elements (before p-value). 

For now, users can use the original matrix and then threshold it by approaches like a fixed threshold and fixed density. I guess we already have something for matrices as well.

ftadel, 2020-11-19:
Thanks for your very prompt reply!

> for the new measures, there is no computation of the p-value, so what you see here is not meaningful. (I thought I made it clear previously but If not that's my fault)

There ARE `pValues` returned by `bst_cohn.m` for `icohere2019` and `lcohere2019`:
https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/connectivity/bst_cohn.m#L358
Maybe this line should be removed?

HosseinShahabi, 2020-11-19:
Sure. As you see I commented no p-value 

https://github.com/brainstorm-tools/brainstorm3/blob/d94cce1d6d3ccbc41b9ad84275cca00ba59952ca/toolbox/connectivity/bst_cohn.m#L351

But probably I added #358 to avoid error, based on the similarity to the previous version. 
The point is that the previous "imaginary coherence" is the squared version of the "lagged coherence 2019", but I wrote the same formula for the p-value. 

So you can first remove the square from here (abs(Gxy).^2) --> (abs(Gxy)) and run it again: 
https://github.com/brainstorm-tools/brainstorm3/blob/d94cce1d6d3ccbc41b9ad84275cca00ba59952ca/toolbox/connectivity/bst_cohn.m#L358

If it was similar to the previous version you can keep it. Else you can remove it. You can do the latter in any way!

ftadel, 2020-11-19:
If your first intention was not to compute any p-value here, let's do without for now, instead of adding something that is not properly investigated:
https://github.com/brainstorm-tools/brainstorm3/commit/3a862a17ebf248feb0db21170c55a6987a2f80b7

This will have to be correctly documented in the tutorial (the fact that the mscohere thresholds at p<0.5, while cohere2019 doesn't).

Top: icohere, Bottom: icohere2019 (different amplitudes but similar shape)
[image]

ftadel, 2020-11-19:
Do you think there is anything that should be added here?
If so, please re-open the issue.

Thanks

closed 2020-11-19 by ftadel