brainstorm-tools/brainstorm3 #368: Feature requestion: Coherency

christian-oreilly, 2020-12-03:
The coherence (also referred to as Magnitude-squared Coherence (MSC)) is derived from the complex-valued coherency measure. In some context, it can be interesting to look at the coherency, for example when someone is interested in studying the functional connectivity as a function of the phase-shift. This measure (coherency) is currently not available to brainstormers. The difference in mathematical definition of coherency, coherence, and imaginary coherence can be found for example here https://mne.tools/dev/generated/mne.connectivity.spectral_connectivity.html It could probably be added to the switch @ https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/connectivity/bst_cohn.m#L232 and support this functionality at relatively little cost, if the fact that the returned values are complex numbers does not cause issues elsewhere (e.g., visualisation functions and such).

ftadel, 2020-12-04:
@HosseinShahabi @sbaillet @richardmleahy 
Is it something something you'd like to distribute?

@HosseinShahabi 
Could you add this to your re-organization of the coherence/connectivity process?

We have ways to handle files with complex values for time-frequency results, I guess similar mechanisms could be used for connectivity results?

HosseinShahabi, 2020-12-07:
Hi @christian-oreilly 

Thanks for your suggestion. I think the implementation of this should not be difficult and it seems that we can handle the complex values. I understand the application of this but as you know we have other measures to assess the phase-difference among signals. 

We try to be a bit conservative about the functions in Brainstorm. It means that we do not include all available metrics in our software unless they are well-known. Would you please provide me with a paper that describes this metric and its importance/application?

Thank you.

christian-oreilly, 2020-12-08:
Well coherency is the underlying transform from which we can compute the "imaginary part of coherency" by taking its imaginary part (as its name indicates) whereas coherence is obtained by taking the magnitude of coherency. But there are other possible ways to derive a real-numbered value from this complex-valued measure than taking the imaginary part or the magnitude. I guess that the question is whether or not you believe it is worth giving the researchers access to the underlying transform rather than to give only access to the these two derived metrics. Since coherency is complexed-valued, it will generally not be used "directly" because it is difficult to display and analyse in a meaningful way complex-valued array without first reducing it to real values. For my part, I am interested in looking at how coherence varies as function of the phase-lag. 

Example of relevant references include:
- https://pubmed.ncbi.nlm.nih.gov/9402881/
- https://pubmed.ncbi.nlm.nih.gov/17418592/

ftadel, 2020-12-08:
If this is not a final measure that can be used or displayed directly, it might not be a good candidate for being added to the Brainstorm interface... 

@christian-oreilly 
Would it help you if we only added one extra possible option for `CohMeasure`, which simply doesn't compute any measure, and if this option was not supported anywhere else? But this is maybe something that you can do yourself, if you exactly know what you want...?
How low-level are you willing to work?

christian-oreilly, 2020-12-08:
I am not sure what you mean by "which simply doesn't compute any measure", it would be of no use if it does not compute the coherency. 

> But this is maybe something that you can do yourself, if you exactly know what you want...?

Sure, no problem. I'll code whatever I need.

ftadel, 2020-12-08:
> I am not sure what you mean by "which simply doesn't compute any measure", it would be of no use if it does not compute the coherency.

I meant "no measure on top of the complex coherency value".

> Sure, no problem. I'll code whatever I need.

If you code (in a clean way) anything that you think could be useful for others, feel free to open a PR to submit it. 
(as long as it doesn't add it to the visible options of coherence process)

ftadel, 2021-03-13:
@christian-oreilly  Do we need to keep this issue open?

christian-oreilly, 2021-03-14:
Not for me. I computed these values using another package since it was the most straightforward solution for my immediate needs.

closed 2021-03-14 by christian-oreilly