brainstorm-tools/brainstorm3 #370: Nirs unit in brainstorm

Edouard2laire, 2020-12-11:
hello, 

I am writing this issue to report a bug on how units are handled with nirs data. Currently, in nirstorm we are dealing with three types of signal: raw signal I (intensity of light detected by the detector, we usually don't specify the physical unit), dOD which is -log(I/Io) where I0 is a normalization factor, typically the mean over the baseline: this is unitless, we then have the concentration calculated after the beer-lambert law in mol/l and all the subunit (note that the typical hemodynamic signal is in mmol/l (noted mmol.l-1)  or micromole/l (noted mumol.l-1or \mumol.l-1 in nirstorn) and that the most common unit used to display nirs is mmol/l but most processed in nirstorm will process and save then in mumol/l) 

Or currently, raw signal is marked as dOD, dOD is rightfully marked as unitless when having the entire signal, but if you import it into the database to perform block-averaging then it is labelled as mmol/l. Concentration is labelled as mol/l but the data displayed are in mmol/l (see the following capture of brainstorm and what we have when exporting data in Matlab, also you can see that the data cursor in brainstorm rightfully indicate that data are in mmol/l but not the ylabel of the figure)

[image]

I can transfer you a subject will the different data calculated so you can do some test.

Best regards, 
Edouard

ftadel, 2020-12-14:
Hmmm... could you directly tell me what needs to be fixed?

ftadel, 2021-03-13:
Ping

ftadel, 2021-09-19:
Is this still an active issue?

ftadel, 2021-10-20:
I also meant: for consistency, use the same notation everywhere for the same units.
Do not use `mol.l-1` somewhere and `mol/l` somewhere else, it adds to the confusion.

> So i don't understand why this information is not used by the function bst_getunits.

`bst_getunits` should be used only when DisplayUnits is not defined.
DisplayUnits was not used when reporting min/max values in the selected time window, as you reported.
I tried to fix that here: https://github.com/brainstorm-tools/brainstorm3/commit/70e2234d4499b2456bae9bdff259891a4b98fe5b
Can you please check that it works for handling the NIRST units?

ftadel, 2021-10-21:
> However, I am not sure if we can simplify Brainstorm functions or if we need to still be compatible with the file created before that fix?

I'm not sure, this depends only on Nirstorm.
In bst_colormaps.m, you could add for instance:
`case {'mol.l-1','mol/l'}, fFactor = 1;`

And then make sure everywhere else that you only use `mol/l` and derivates, which are more readable?

> Here the min/max doesn't take into account the scale factor (10^-5) because bst_getunits is not called.

Indeed, this is designed to use either a fixed unit OR something that Brainstorm guesses, not a mix of both.
Do you really need to set `DisplayUnits` everywhere?
Is it not possible to guess the units of the NIRS signals from their context + amplitude range?

> btw, would it be possible to change the location of the scale factor above the vertical axis as we can't see it if there is an event marked at the begening of the file, for exemple transient event from filtering ?

This is defined by Matlab, I think... I'm not sure what you can do about it.
In time series figures, we designed everything so that there no additional scaling factor for MEG or EEG.
The best option is to have the values properly scaled along the Y axis, with YTicks that never require to add any power of tens multipliers... And therefore, no fixed units OR DisplayUnits that are set in order to have Y values always between 0.001 and 1000.

closed 2021-11-05 by Edouard2laire