brainstorm-tools/brainstorm3 #373: Problem with loading EDF files - Function "in_fopen_edf.m"

HosseinShahabi, 2021-01-08:
Hi Team, 

I have been using the function "in_fopen_edf.m" for a long time. Recently, I noticed It cannot load some EDF files. I checked the recent update (01/07/21) but it couldn't help. 

I guess probably for some files the header is not properly set. They all come from the same source and the same subject but it only happens in some of them. 

The first problem is here: 

https://github.com/brainstorm-tools/brainstorm3/blob/eb925ea63a11293e5b0cd28789fad01f14967e12/toolbox/io/in_fopen_edf.m#L372

It runs other functions until it reaches the following line. 

https://github.com/brainstorm-tools/brainstorm3/blob/eb925ea63a11293e5b0cd28789fad01f14967e12/toolbox/io/in_fread_edf.m#L163

The variable "offsetStart" is a vector instead of a scalar since "offsetHeader" is a vector. 

https://github.com/brainstorm-tools/brainstorm3/blob/eb925ea63a11293e5b0cd28789fad01f14967e12/toolbox/io/in_fread_edf.m#L144

So this should be related to that specific file but there must be a way to solve it. I tried to only select the first element of that vector and the error was gone but I'm not sure it is a correct way to handle it. 

sFile.header.hdrlen = sFile.header.hdrlen(1) ;

After that, there is another error in 

https://github.com/brainstorm-tools/brainstorm3/blob/eb925ea63a11293e5b0cd28789fad01f14967e12/toolbox/io/in_fopen_edf.m#L395

Similarly, "t0_rec" is a vector so two sides of this logical expression will have different lengths and it causes the error. I tried to understand the logic behind the code but it was difficult and I'm not very familiar with EDF structure. especially there were weird annotations in those files: 

K>> splitAnnot{1}
ans =
    ';ÿ[þ¿ý ýLÄß¶ºJ fÿÙþvþmþþLÿy ºÈiÒö÷k3ÿÿ:ÿ²þjþsþ´þ¦ÿÁ âj»ØÊä¬ ÿøþþbþyþðþ'

I would appreciate it if we can fix it very soon.

ftadel, 2021-01-10:
I don't understand how what you reported here can happen. From in_fopen_edf:
```
hdr.hdrlen = str2double(fread(fid, [1 8], '*char'))
```
`hdrlen` can't be a vector, since str2double only returns scalars.

The following doesn't make sense either, unless the annotation channel is corrupted:
```
K>> splitAnnot{1}
ans =
';ÿ[þ¿ý ý��L�����Ä�ß�¶�º���J fÿÙþvþmþ�þLÿy º�È�i�Ò�ö�÷���k�3�ÿÿ:ÿ²þjþsþ´þ¦ÿÁ ��â�j�»�Ø�Ê���ä�¬ �ÿøþ�þbþyþðþ'
```

Please:
1) Make sure you don't have any other function interfering with Brainstorm/Matlab in your Matlab path
2) Make sure that you are using the latest version of the EDF reader (I updated it a few days ago - see the forum)
3) Share an example file so I can reproduce and fix these errors

HosseinShahabi, 2021-01-19:
Thank you, Francois. 

I updated my Brainstorm to the last version after your changes to the EDF reader. 
However, I realized the EDF files were corrupted at the source because of some modifications, and this function cannot reverse that process. Fortunately, it was solved by sending new files. 

I close this Issue.

closed 2021-01-19 by HosseinShahabi