brainstorm-tools/brainstorm3 #374: The two versions of imaginary coherence

ShenghongHe, 2021-01-12:
Hi, could anybody please briefly explain the difference between the two versions of imaginary coherence in this code for me? I could only find references for the new version (abs(imag(C))) but not for the old version (before 2019). I would like to use the old version because it gave me better results in my case. Any references related to the old version of the imaginary coherence would be very helpful. Thanks a lot! -Shenghong

ftadel, 2021-01-12:
I meant the user forum: https://neuroimage.usc.edu/forums/
The github issues are dedicated to fixing bugs in the software, and not designed to be explored by other users looking for information. But anyway­... we will answer here and then close the issue. Please use the forum for future questions - thanks.

I think there are a lot more references for the "new" code, and this is why it was modified.
I don't have any reference to cite for the "old" code, but maybe @HosseinShahabi  or @richardmleahy do?

In general, it is not a good solution to chose a method over another because "it looks better".
You should design your analysis based on the literature existing in your field, and stick as close as possible to the documented processing pipelines you refer to, in order to avoid multiplying the possible confounds.
The Brainstorm signal processing specialists recommended changing the default method for the coherence computation. You can can reject this recommendation, but then we will probably not be able to help you justifying this choice.

ShenghongHe, 2021-01-12:
Dear @ftadel, 
I apologize again for posting the questions here and thank you vey much for still answering my question here. Next time I promise the question will on the user forum :)

I agree with you it should be better to choose a solution based on the research question itself rather than the results. I am trying to quantify PPN-cortical coherence during gait. Because of the potential movement artefact during gait, imaginary coherence maybe a better solution than the more normal magnitude-squared coherence. I found the nice code here in brainstorm3 by accident and found the results were exactly as I expected. So quite curious to understand a bit more on this. It would be great if @HosseinShahabi or @richardmleahy could help to clarify for me, otherwise please feel free to close this issue.

Many thanks,
Shenghong

ShenghongHe, 2021-01-12:
I just realized that when the coherence was quantified based on wavelet transformation rather than FFT, the results for these two versions of imaginary coherence were very similar. Only the scale for the old version was about 10 times smaller than the new version. But the results of both versions looked much better than magnitude-squared coherence, which gave me quite big values in high frequency band which seemed unnormal. Thanks a gain!

HosseinShahabi, 2021-01-12:
Hi @ShenghongHe 

As @ftadel said, please use the forum for these questions in the future. 

If you check the description for GUI, you will find this 

%     ============ 'icohere2019', 'lcohere2019' =============
%     Imaginary Coherence (IC)          : abs(imag(C))               
%     Lagged Coherence (LC)             : abs(imag(C))/sqrt(1-real(C)^2)
%
%     ========= 'icohere' (before 2019) =========
%     Imaginary Coherence (IC)          : imag(C)^2 / (1-real(C)^2)

So, basically (IC before 2019) is the squared version of the (LC 2019). We made this change to make the definition and naming consistent with the literature. 

If you use lagged-coherence (2019) you must get a similar results as old IC (except the square). Please read the following article which completely discusses these measures. For some reason that you find in the paper, the lagged-coherence have some merits when comparing with traditional imaginary coherence (IC 2019 in Brainstorm)

https://arxiv.org/pdf/0706.1776.pdf

@ftadel I added this paper as a reference to the connectivity tutorial. 

Thank you,
Hossein

ShenghongHe, 2021-01-12:
Hi @HosseinShahabi,

Thanks a lot for your clarify. Sure no problem, next time I will make sure to use forum for questions. Yes, this reference seems to be very clear on this issue. Thanks for sharing it! So theoretically the old version is not a 'real' imaginary coherence because it still relied on the real part of the coherence, but as you said, a squared lagged coherence.

Thank you all for your help on this!

All the best,
Shenghong

closed 2021-01-12 by ftadel