brainstorm-tools/brainstorm3 #379: issue in mri_getlabels causing crash

ajoshiusc, 2021-02-01:
brainstorm3/toolbox/anatomy/mri_getlabels.m

The following code in the mri_getlabels needs to be updated
```
if (nargin < 3) || isempty(isForced)
    isForced = [];
end

```
This causes the problem at the end of the function at the line

`if isempty(AtlasName) && ~isForced
`
the isForced cannot be empty.

The block above should be (possibly) changed as follows:
```
if (nargin < 3) || isempty(isForced)
    isForced = false;
end

```

ftadel, 2021-02-02:
I had fixed it on my computer but not pushed it yet, as it is mixed with other work in progress.
I push some of it here: https://github.com/brainstorm-tools/brainstorm3/commit/6172b6c8204be99eb232ad1c443904a474f301e2
I hope the commit will fix the bug without causing other bad side effects...

Thanks for reporting it!

closed 2021-02-02 by ftadel

tmedani, 2021-02-05:
hi @ftadel 

with the last update, when I try to import the MRI to a subject I got this error:

***************************************************************************
** Error: Line 160: Undefined function 'mri_getlabels' for input arguments of type 'struct'.
** 
** Call stack:
** >import_mri.m at 160
** >bst_call.m at 28
** >tree_callbacks.m>@(h,ev)bst_call(@import_mri,iSubject,[],[],1) at 562
** 
***************************************************************************

I just skipped it right now and comment on these lines for my case, without further investigations.

ftadel, 2021-02-05:
`mri_getlabels.m` is in the distribution:
https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/anatomy/mri_getlabels.m

Maybe you updated files separately... Make sure you merge all the last changes, or simply download a new version from the Brainstorm website.