brainstorm-tools/brainstorm3 #381: MRI Coordinates values on the bst MRI viewer

tmedani, 2021-02-10:
Hi @ftadel 

I have just noticed that the MRI coordinate displayed on the top of the MRI are different from those displayed on the table. 

[image]

It seems that the values on the top are more correct. 

best,

rcassani, 2021-02-10:
Hi @tmedani, I'm using the 08-Feb-2021 version and the coordinates in the figures and the table agree in the MRI Viewer.

ftadel, 2021-02-10:
Above the slices: this is the slice index in the volume.
In the Coordinates panel, this are coordinates in millimeters.
You have the same values only if the voxel size is 1x1x1mm.

closed 2021-02-10 by ftadel

tmedani, 2021-02-10:
Thanks, @ftadel  & @rcassani,

Indeed, my MRI is not wired in this example, I miss to check it.

thank you