brainstorm-tools/brainstorm3 #384: request new feature/improvement : display dipoles orientation on the MRI viewer

tmedani, 2021-02-17:
Hi @ftadel & all,

I'm just wondering if it's doable to add the display of the full dipole on the MRI viewer, 

This option is already possible on the MRI 3D viewer, and there is a panel to control the parameters from the GUI. 

The current version of the MRI viewer displays only the position of the dipole, can we add the orientation?

If yes, I can help at some points.

This is the MRI viewer :
[image]

 here is it on the 3D viewer.
[image]

Also, it will be good if the MRI slices on the 3D viewer can be aligned (centered) to the currently selected dipole position,

for example by using the shortcut keyboard  "M" as in the MRI viewer.

@martcous @rcassani 

Thanks :)

ftadel, 2021-02-18:
> The current version of the MRI viewer displays only the position of the dipole, can we add the orientation?

There is some misunderstanding here: the "MRI Viewer" view of the dipoles is not a view of the dipoles, but a summary of the dipoles density, useful when there are hundreds of localized spikes for instance.  This was never documented properly...
At the moment, there is no display of single dipoles coded for the MRI viewer. This is doable, but is represents too much work for me to add it to my agenda of the new few months, sorry.
You can use the orientation shortcut from 3D views (keys 1 to 6) to get only one plane at a time, get produce figures that are more similar to the MRI viewer.

> Also, it will be good if the MRI slices on the 3D viewer can be aligned (centered) to the currently selected dipole position,

Done. I added a button "View/MRI" button in the dipole info window.
https://github.com/brainstorm-tools/brainstorm3/commit/5e477b526d2561f8bd487cc3e32fdbd22fa542c6

[image]

closed 2021-02-18 by ftadel

tmedani, 2021-02-19:
Thank you Francois for the new adding and the clarification. 

What you recommend is already what I'm using, but I found it slightly not intuitive and though it's maybe better to have the three-view on the same screen. 

I will put it on my to-do list, maybe I can work on it when it's possible.

Thanks & A+