brainstorm-tools/brainstorm3 #385: custom magnetometer device in ctf format applying unwanted "eps" gain

guhandi, 2021-03-01:
Hi,

I work for a startup where we are building a custom MEG system. I am writing a converter from our data format to **ctf** format so it can be imported into Brainstorm easily. I want to set the "_SensorType_" of the device to 4 because we want our sensors to be magnetometers (according to the CTF documentation and attached code snippet from "_ctf_read_res4.m_" file). However there is no condition in the reader to deal with SensorType = 4, resulting in an arbitrary eps gain being applied to my data which makes it unusable.

[image]
[image]

For a short term fix i found that if i replace line 475 in ctf_read_res4.m from:
`gain_chan(gain_chan == 0) = eps;`
To:
`gain_chan(gain_chan == 0) = 1;`
     
Then the data gets imported as intended.

[image]

For a longer term solution it could potentially be very useful to have our own device listed as an import type as well so we can better align our framework with more compatible data importing functions. But the quick fix should allow us to atleast get the data collected from our system into brainstorm in a usable way!

ftadel, 2021-03-02:
Indeed, there are no magnetometers in the CTF MEG.  However, I'm not sure what are all the implications of replacing zeros with eps.
This function has been untouched and stable for a very long time. It is used directly by many MEG centers with CTF systems, and I'm very reluctant to change anything that I don't have a full control over, especially if this is not related with the CTF file format itself.

As you mentioned, if you need to modify the format for the MEG system you develop, the correct approach would be to write a new set of readers (in_fopen_...m and in_fread_...m functions). Branching from an existing file format would offer no benefit if you can't reuse directly the software capable of processing this file format. 
If you are interested, I can help you with this. When you are ready for this, please post a message on the user forum (category Request), or open directly a PR on this repo if you don't need help with it.

A format you could re-use without alterations is maybe the FIF format. Using a format you do not need to alter would allow a direct compatibility with all the programs already capable of processing it. 

In both cases, you'd need to add definitions of your sensors to the coil_def.dat file, shared between Brainstorm and MNE:
https://neuroimage.usc.edu/forums/t/edit-channel-coil-def-dat/19346

closed 2021-03-02 by ftadel

guhandi, 2021-03-05:
Hey @ftadel I'm trying the approach to write our data into the FIF format but am finding it difficult to find a place to start. Would you be able to point me to an example or some clear documentation on how to write FIFs? Ive seen this PDF but its not the easiest to get started with. 

[Fiff.pdf](https://github.com/brainstorm-tools/brainstorm3/files/6087651/Fiff.pdf)

juangpc, 2021-03-05:
Hi, 

If you have any particular question or feel stuck, you can also let me know. 

Good luck,

ftadel, 2021-03-05:
> Would you be able to point me to an example or some clear documentation on how to write FIFs? 

Sorry, this is something we've never done ourselves, the FIF format being a bit cumbersome.
Check with MNE-C or MNE-Python.

The BrainVision format could also be an option:
https://neuroimage.usc.edu/forums/t/creating-matlab-raw-data-file/24904

guhandi, 2021-03-24:
@ftadel
Hi,
Ive been working with FIFs and this seems to work. Quick question - I notice the fiff format uses the Neuromag RPA coordinate system and Brainstorm uses SCS/CTF. The transform from RPA->CTF is only [x,y,z] = [y,-x,z]. 

Is there an easy way to have brainstorm deal with difference during read time?

ftadel, 2021-03-25:
> The transform from RPA->CTF is only [x,y,z] = [y,-x,z].

No, this is not that straightforward.
CTF: Origin = midway on the line joining LPA and RPA. 
Neuromag: Origin = Intersection of the line L through LPA and RPA and a plane orthogonal to L and passing through the nasion.

See: https://neuroimage.usc.edu/brainstorm/CoordinateSystems

> Is there an easy way to have brainstorm deal with difference during read time?

Brainstorm converts anything to SCS/CTF when importing data, based on the NAS/LPA/RPA positions. The transformation is kept in the channel file, field TransfMeg:
https://neuroimage.usc.edu/brainstorm/Tutorials/ChannelFile#On_the_hard_drive