brainstorm-tools/brainstorm3 #387: E-phys: Integrate plugin manager and fix code online tutorials

ftadel, 2021-03-13:
@mpompolas @sbaillet : 

# 1. Plugin manager

I added a new solution for managing the packages downloaded by Brainstorm:
https://neuroimage.usc.edu/brainstorm/Tutorials/Plugins

I updated everything in Brainstorm but the functions of the ephys toolbox, since I can't test them.
To finish this upgrade, we need to defined as plugins (ie. add an entry in `bst_plugin.m` and replace the `DownloadFile` calls with `bst_plugin('Install', ...)`) :
- DeriveLFP: https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/process/functions/process_convert_raw_to_lfp.m#L112
- Kilosort: https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/process/functions/process_spikesorting_kilosort.m#L565
- KilosortWrapper:  https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/process/functions/process_spikesorting_kilosort.m#L590
- Phy:  https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/process/functions/process_spikesorting_kilosort.m#L616
- npy-matlab: https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/process/functions/process_spikesorting_kilosort.m#L637
- UltraMegaSort: https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/process/functions/process_spikesorting_ultramegasort2000.m#L110
- WavClus: https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/process/functions/process_spikesorting_waveclus.m#L104

# 2. File formats

I've updated the file readers, but can't test them, either because I don't have any example data file (TDT, Plexon), or because :
- **NWB**: Done but needs to be tested - none of the example files publicly distributed work with these readers
https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/io/in_fopen_nwb.m#L30
- **TDT**: Done but needs to be tested - I don't have any example file
https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/io/in_fopen_tdt.m#L33
- **Plexon**: Automatic download of reader could be added: 
https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/io/in_fopen_plexon.m

Additionally, an issue was reported on the forum about the Plexon .pl2 file format:
https://neuroimage.usc.edu/forums/t/problems-of-import-a-pl2-file/26107

# 3. Errors in the tutorials

There is a number of serious issues related the online e-phys tutorial, which are probably one of the cause of this toolbox not being used at all. It has not been cited since its publication, and we have no forum posts about it. 
I posted a review in early 2019, I'm not sure any of it was ever addressed:  
https://github.com/brainstorm-tools/brainstorm3/issues/170

# 4. Action plan?

If we don't address these few (minor) issues, this impressive amount of work will never be used or cited.
I can help, but I can't lead this initiative, since I don't have any example file of expertise in any of the fields involved, and I didn't write any of the code involved.
Please let me know how you would like to proceed.

ftadel, 2021-09-19:
@sbaillet @mpompolas Any plans for solving these issues?

mpompolas, 2022-01-14:
Started working on this

mpompolas, 2022-04-06:
### 1. Plugin manager
Added everything on the plugin manager

### 2. File Formats

### Plexon (.plx, .pl2)
[image]

### NWB
Tried it on Public datasets on DANDI: https://gui.dandiarchive.org/#/dandiset/000003/0.210812.1448/files?location=sub-YutaMouse37%2F
[image]

### TDT
Added support for reading Synapse files for user-friendly event labeling.
[image]

### Intan (.rhd, .rhs)
[image]

### 3. Errors in the tutorials
Most issues have been addressed. What remains, and would need engagement from @ftadel for points 1-4.
1. Deriving LFP signals from raw signals / convert function lsqnonlin.m to Octave.
2. Function [Spike triggered average](https://neuroimage.usc.edu/brainstorm/e-phys/functions#Spike_triggered_Average_-_DRAFT_.28CONVERSION_OF_CHANNELS_TO_SEEG_NEEDED.29) needs SEEG channel types. The function `Consider as SEEG/ECoG` doesn't assign different types due to the non-standardized channel labels on the tutorial file.
3. I supply the MRI in .nii format, and the tessellated cortical surface in BrainVisa format (.mesh) with the .zip. These are completely optional for the tutorial. We can skip them. Also, the co-registration is a bit off:
[image]

4. Add the dataset from my OneDrive to the USC server.
5. Check all the progress bar popups (WIP, will update/remove from here once it's done)
6. 3. GPU on Kilosort (WIP, will update/remove from here once it's done)

### 4. Bug reports
Everything has been covered. Some are waiting responses from the users.

ftadel, 2022-04-25:
This is all addressed in PR https://github.com/brainstorm-tools/brainstorm3/pull/516

closed 2022-04-25 by ftadel