brainstorm-tools/brainstorm3 #388: NIRSTORM: Move to the plugin manager?

ftadel, 2021-03-13:
@multifunkim @thomas-vincent @Edouard2laire

I added a new solution for managing the packages downloaded by Brainstorm:
- Tutorial: https://neuroimage.usc.edu/brainstorm/Tutorials/Plugins
- Commit: https://github.com/brainstorm-tools/brainstorm3/commit/a8e0a54ef97f1658a2d268e211f586b87912f22a

It would be great if we could also move NIRSTORM to this framework, in order to have everything managed in a centralized way. This will also allow to have automated builds including all the dependencies (to keep on having NIRSTORM included in the compiled distribution of Brainstorm).

Who could help me with this task?
Maybe we could organize a short meeting to plan these changes?

ftadel, 2021-04-27:
I've just recompiled Brainstorm with the updated BrainEntropy and Nirstorm code.
Nirstorm should now work with both with the Matlab and the compiled distributions of Brainstorm.
Please test it and confirm that we can close this issue.

ftadel, 2021-09-02:
At this moment, Nirstorm is still excluded from the compiled version of Brainstorm.
It will be the case until this issue gets resolved: https://github.com/Nirstorm/nirstorm/issues/155

Let me know when I should try include to recompile Brainstorm with Nirstorm.

Edouard2laire, 2021-09-02:
Yes, i am still working on making the replacement process for the problematic ones. I should be able to merge them soon with the main branch of Nirstorm. I will let you know when its done :)

ftadel, 2021-09-21:
Nirstorm repo issues were closed: https://github.com/Nirstorm/nirstorm/issues/155

Nirstorm is restored as a compiled plugin: https://github.com/brainstorm-tools/brainstorm3/commit/0428eb6da54d4a4a7c575aaca4bb88d2fd5c0f5e

closed 2021-09-21 by ftadel