brainstorm-tools/brainstorm3 #389: BrainEntropy: Move to the plugin manager?

ftadel, 2021-03-13:
@multifunkim @aliobaibk @YounesZ @klacourse

I added a new solution for managing the packages downloaded by Brainstorm:
- Tutorial: https://neuroimage.usc.edu/brainstorm/Tutorials/Plugins
- Commit: https://github.com/brainstorm-tools/brainstorm3/commit/a8e0a54ef97f1658a2d268e211f586b87912f22a

It would be great if we could also move the BrainEntropy toolbox to this framework, in order to have everything managed in a centralized way. This will also allow to have automated builds including all the dependencies, to have the BE toolbox compiled with Brainstorm.
Having the MEM included in the compiled distribution of Brainstorm has been requested recently on the forum:
https://neuroimage.usc.edu/forums/t/best-toolbox-in-brainstorm-compiled-version/25898

Who could help me with this task?
Maybe we could organize a short meeting to plan these changes?

Edouard2laire, 2021-03-15:
Hello, 

I'll raise the question during our MEM meeting tomorrow morning.  Personnally, i won't have time before early april. 

Édouard

ftadel, 2021-04-01:
I copy here the list of other requests I posted on the forum:
https://neuroimage.usc.edu/forums/t/issue-trying-to-compute-surces-wiht-mem-with-eeg-recordings/26910/2?u=francois

- Address the issues reported on the forum, for example: 
https://neuroimage.usc.edu/forums/t/can-not-compute-sources-using-mem/22678/6
https://neuroimage.usc.edu/forums/t/issue-trying-to-compute-surces-wiht-mem-with-eeg-recordings/26910
- Fix the EEG/Epilepsy tutorial: 
https://neuroimage.usc.edu/brainstorm/Tutorials/Epilepsy#Source_analysis:_MEM_inverse_solution
- Fix the BEst tutorial:
https://neuroimage.usc.edu/brainstorm/Tutorials/TutBEst
- Address the questions about the compilation of your functions together with Brainstorm: 
https://neuroimage.usc.edu/forums/t/best-toolbox-in-brainstorm-compiled-version/25898/3
- Optionally (needed for compiling with Brainstorm): Remove the dependencies to Matlab toolboxes other than the Signal Processing Tb (exemple: find an alternative to fminunc)
- Standardize the way your method is called: relying on detecting a file named "baseline" in a folder is not a robust solution, instead please consider using the Process2 tab, with the baseline on one side and the target data on the other (similarly to the Baseline normalization or Epileptogenicity processes)

aliobaibk, 2021-04-02:
Hello @ftadel 

- Got it, I will be updating BEst for better handling the baseline file in the forthcoming days and @multifunkim is working on the tutorials.
- Concerning the dependence to `fminunc`, the toolbox defaults to `minFunc` (shipped with BEst) if `license('test', 'Optimization_Toolbox')` returns false.

ftadel, 2021-04-05:
> Concerning the dependence to fminunc, the toolbox defaults to minFunc (shipped with BEst) if license('test', 'Optimization_Toolbox') returns false.

Wouldn't there be Matlab-only alternatives for the optimization functions you need?
Maintaining MacOS mex files over the years can be painful.
Including them in the compiled version of Brainstorm might be complicated too.

aliobaibk, 2021-04-05:
Hello @ftadel 

`minFunc` is in fact a MATLAB-only alternative but uses by default MEX files for speeding things up.
I will be adding a parameter to the BEst panel so that users can choose to use or not this speedup feature (users can already choose between `fminunc` and `minFunc` with `mincFunc` being automatically set if the optimization toolbox is not present).

Thanks

ftadel, 2021-04-06:
> Concerning the dependence to fminunc, the toolbox defaults to minFunc (shipped with BEst) if license('test', 'Optimization_Toolbox') returns false.

This test doesn't work: it tests for what the license allows, not for what is installed on the computer.
See what I get in my MATLAB install, which does not include the optimization tb:

```
>> license('test', 'Optimization_Toolbox')

ans =

     1

>> ver
-----------------------------------------------------------------------------------------------------
MATLAB Version: 9.10.0.1602886 (R2021a)
MATLAB License Number: 521249
Operating System: Microsoft Windows 10 Pro Version 10.0 (Build 19042)
Java Version: Java 1.8.0_202-b08 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
-----------------------------------------------------------------------------------------------------
MATLAB                                                Version 9.10        (R2021a)
Curve Fitting Toolbox                                 Version 3.5.13      (R2021a)
Image Processing Toolbox                              Version 11.3        (R2021a)
MATLAB Compiler                                       Version 8.2         (R2021a)
MATLAB Compiler SDK                                   Version 6.10        (R2021a)
Parallel Computing Toolbox                            Version 7.4         (R2021a)
Signal Processing Toolbox                             Version 8.6         (R2021a)
Statistics and Machine Learning Toolbox               Version 12.1        (R2021a)
Wavelet Toolbox                                       Version 5.6         (R2021a)
```

You should replace this test with a much simpler:
`if exist('fminunc', 'file')`

aliobaibk, 2021-04-06:
Hello @ftadel

Indeed, as you note, one may have a license for a toolbox but not have it installed and the test that you propose catches partially this case, and I agree that it should be added.
In addition, one may have a toolbox installed with no license to run its functions and the previous test with `license(...)` catches this case.
I will be combining the two.

Thanks

ftadel, 2021-04-27:
It should be working now, both with the Matlab and the compiled versions of Brainstorm.
Please test it and confirm that we can close this issue.

The work left to do includes only documentation:
- the EEG/Epilepsy tutorial:
https://neuroimage.usc.edu/brainstorm/Tutorials/Epilepsy#Source_analysis:_MEM_inverse_solution
- the BEst tutorial:
https://neuroimage.usc.edu/brainstorm/Tutorials/TutBEst

aliobaibk, 2021-04-28:
Hello @ftadel,
The tests were successful. 
@multifunkim is currently working on the tutorials.
Thanks

closed 2021-04-29 by ftadel