brainstorm-tools/brainstorm3 #413: I used a coregistered T1 and T2 to create a FEM mesh with SIMnibs ,and the segmentation seemed to be ok  but the FEM generated isn't aligned with the T1 in the database anymore so  I'm wondering if the problem is solved already ?

tmedani, 2021-05-26:
I used a coregistered T1 and T2 to create a FEM mesh with SIMnibs ,and the segmentation seemed to be ok  but the FEM generated isn't aligned with the T1 in the database anymore so  I'm wondering if the problem is solved already ? 
What can I do ? I tried also brain2mesh but it gave me errors in the segmentation and in the creation of the surfaces.
[image]
Thank You for the support

_Originally posted by @dunjadudu in https://github.com/brainstorm-tools/bst-duneuro/issues/7#issuecomment-833671554_

tmedani, 2021-05-26:
Hi all, 

I'm moving this [issue](https://github.com/brainstorm-tools/bst-duneuro/issues/7#issuecomment-833671554) here, it was reported by @[dunjadudu ](https://github.com/dunjadudu)in the bst-DUNEuro repo, but it's not related to DUNEuro.

ftadel, 2021-05-31:
I'm not sure I understand what the status is.

@dunjadudu 
- What version of SimNIBS have you used exactly? (3.2.?)
- Is the issue that it works with T1, but not with T1+T2?
- Could you please share the SimNIBS output folder (the one with the problem)? 
    - Make a copy of the folder and delete all the files not needed for Brainstorm (to make the download package smaller), and keep only the following files:
    - *T1.nii / *T2.nii
    - *_T1fs_conform.nii.gz / *_T2fs_conform.nii.gz
    - *_final_contr.nii.gz
    - *.msh
    - The entire subfolder: `m2m_subjectid/segment/cat/`
- Zip the folder, upload the folder somewhere and post the download link here
- If you can't share this publicly, please sent it as private messages or via email (francois.tadel@mcgill.ca, medani@usc.edu)

@tmedani 
- Is that version of SimNIBS supposed to be supported?
- Is it worth it investing time fixing this, or should we switch directly to SimNIBS 4 / CHARM?

tmedani, 2021-06-02:
Hi @ftadel 

Thanks for the follow-up, I'm not sure but I think this error can be related to some missing data on the MRI, but we need to check it.  Let's check the data as soon as we received it.

Regarding CHARM, it's still not distributed within the last release of SIMNIBS, 
as soon as they release it we may switch to it.

dunjadudu, 2021-06-04:
Hi all, sorry for the late answer, I tried using nifti files and the resulting segmentation/cortex reconstructed is well coregistered so I think I  should not use MRI after freesurfer processing ( old version probably ) .Thank You for the support.
According to the next step (DUNeuro settings) I'm interested at trying to use FEM and MEG data( ELEKTA) . The error that I found is in the attachment , I've already tried thr constrain option on the grey matter ( I tried using only grey matter , forcing the dipoles to the grey matter) .

Ing. Dunja Duran

Biomedical Engineering Unit

Dept. of Neurophysiology

Fondazione IRCCS Istituto Neurologico "C. Besta”

via Celoria 11, 20133 Milano, Italia
tel. +39 0223942958, 0223942166

***@***.***

[IRCCS Besta 5x1000]<https://www.istituto-besta.it/display-page/-/asset_publisher/zbY5D3OWMnJU/content/nuova-campagna-5-per-mille-abbiamo-bisogno-di-una-mano-la-tua>

Informativa breve resa ai sensi dell’art. 13 del Regolamento UE 2016/679
Il titolare del trattamento, Fondazione IRCCS Istituto Neurologico “Carlo Besta” (C.F. 01668320151 – P.IVA 04376340156), con sede in Via Celoria 11 – 20133 Milano (tel. 02.23941 – fax 02.70638217 – PEC: ***@***.***), tratta i Vostri dati in conformità alla normativa vigente in materia di protezione dei dati personali, come meglio descritto nell’informativa estesa reperibile presso la sede della Fondazione, e consultabile anche sul sito istituzionale www.istituto-besta.it. Il Responsabile della protezione dei dati (DPO) è contattabile all’indirizzo di posta elettronica ***@***.******@***.***>

tmedani, 2021-06-04:
@dunjadudu, thanks for the feedback, freesurfer probably changes some information on the MRI data that are used by simnibs.

regarding the second error, could you give more information, there is no attachment in your post.

ftadel, 2021-06-07:
@dunjadudu 

>  I tried using nifti files and the resulting segmentation/cortex reconstructed is well coregistered so I think I  should not use MRI after freesurfer processing ( old version probably ) .Thank You for the support.

I will therefore close this issue.
If you need to add anything to it, feel free to reopen it.

> The error that I found is in the attachment , I've already tried thr constrain option on the grey matter ( I tried using only grey matter , forcing the dipoles to the grey matter) .

Please document your issue in more detailed way on the user forum.
In order to help you and fix possible bugs, we need enough information to reproduce them.
https://neuroimage.usc.edu/forums/

closed 2021-06-07 by ftadel