brainstorm-tools/brainstorm3 #417: [Plugin-system]Usage of the different plugin in brainstorn

Edouard2laire, 2021-06-17:
Hello @ftadel, 

Earlier today, I was arguing with a colleague about the fact that either NIRSTORM was used only in our lab or on a more global scale. Or we don't currently have any value to know how many people are using our toolbox. 

I was therefore wondering if it would be possible to integrate this measure in the plugin system? 

Regards, 
Edouard

ftadel, 2021-06-18:
Agreed.

I added a PHP script on the website to log at least the installations of the plugins:
https://github.com/brainstorm-tools/brainstorm3/commit/0a7e97b49c6b6a99d2cbbe24494a0355067abbd3

To log an event "install" of the plugin "nirstorm" (both strings are free-form, you can log whatever you want):
```https://neuroimage.usc.edu/bst/pluglog.php?c=K8Yda7B&plugname=nirstorm&action=install```

To get the event logs, add `&list=1`:
```https://neuroimage.usc.edu/bst/pluglog.php?c=K8Yda7B&plugname=nirstorm&action=install&list=1```

We don't store anything but the date and event type. No IP or username, because then it gets complicated with GDPR...

If you want to add some custom logs to any NIRSTORM function you can simply add it anywhere in your code:
```bst_webread(['http://neuroimage.usc.edu/bst/pluglog.php?c=K8Yda7B&plugname=' PlugDesc.Name '&action=youraction']);```

closed 2021-06-18 by ftadel

Edouard2laire, 2021-06-18:
Thx a lot Francois. 

I guess, there i nos way too retroactively know how many, currently have Nirstorm installed ?

sbaillet, 2021-06-18:
Keeping track of the NIRS papers that cite Brainstorm is one objective measure of successful adoption.

ftadel, 2021-06-20:
> I guess, there i no way to retroactively know how many, currently have Nirstorm installed ?

I'm not sure how you would like to do that.
Maybe GitHub offers some hidden feature to count the downloads?

> Keeping track of the NIRS papers that cite Brainstorm is one objective measure of successful adoption.

Indeed, you can check the list of publications using Brainstorm, screening for the ones using fNIRS.
https://neuroimage.usc.edu/brainstorm/Pub

One example I received today:
https://www.spiedigitallibrary.org/journals/neurophotonics/volume-8/issue-02/025012/Guiding-functional-near-infrared-spectroscopy-optode-layout-design-using-individual/10.1117/1.NPh.8.2.025012.full#r27