brainstorm-tools/brainstorm3 #419: (spectral) Granger causality implementation

danielemarinazzo, 2021-06-29:
Hello

the traditional way of implementing GC (i.e. comparing a full and a reduced model), leads to some issues, pointed out to in this paper https://www.pnas.org/content/114/34/E7063.short. The main issue is that the reduced model is VAR, the full model is VARMA; the full model has a given model order, the reduced model has order infinite by definition.
Luckily for the field, these issues had been already addressed prior to the Stokes&Purdon paper. One efficient way to solve them is by means of state space models.
More info (and pointers to code) here

https://f1000research.com/articles/6-1710
https://pubmed.ncbi.nlm.nih.gov/29883736/

I would be happy to help, I am not very familiar with the Brainstorm way of coding and dealing with data.

ftadel, 2022-05-25:
Is this issue the same as what is discussed in this PR: https://github.com/brainstorm-tools/brainstorm3/pull/433 ?

danielemarinazzo, 2022-05-25:
yes

rcassani, 2025-08-26:
Closed by #433