brainstorm-tools/brainstorm3 #421: Error when Importing T2

tmedani, 2021-06-30:
Hi @ftadel 

This error does not appear in the previous version of Bst, but it's happenning now when we import the T2 and try to coregister to the T1 already imported to the subject. 

[image]

ftadel, 2021-07-05:
I can't reproduce this error...
Can you please make sure you are testing this with the current `master` version of Brainstorm?
Can you please tell me exactly what you are doing to get this error, from an empty subject, using the FEM tutorial dataset you sent me?

tmedani, 2021-07-05:
Oui, 
same data ==> import MRI, select T2, ==> coregister with SPM==> reslice: Yes, ==> 

`
------------------------------------------------------------------------
Running job #1
------------------------------------------------------------------------
Running 'Coregister: Estimate & Reslice'

SPM12: spm_coreg (v6435)                           16:01:22 - 05/07/2021
========================================================================
Warning: You have used 'state' to read the internal state of a uniform random number generator that is not currently active, which most likely returns a result that is
not valid.  This syntax is not recommended.  See Replace Discouraged Syntaxes of rand and randn to use RNG to replace the old syntax. 
> In spm_coreg>loaduint8 (line 267)
In spm_coreg (line 127)
In spm_run_coreg (line 22)
In cfg_run_cm (line 29)
In cfg_util>local_runcj (line 1688)
In cfg_util (line 959)
In spm_jobman>fill_run_job (line 469)
In spm_jobman (line 247)
In mri_coregister (line 156)
In import_mri (line 290)
In bst_call (line 28)
In tree_callbacks>@(h,ev)bst_call(@import_mri,iSubject,[],[],1) (line 562) 
Warning: Using 'state' to set RAND's internal state causes RAND, RANDI, and RANDN to use legacy random number generators.  This syntax is not recommended.  See Replace
Discouraged Syntaxes of rand and randn to use RNG to replace the old syntax. 
> In spm_coreg>loaduint8 (line 268)
In spm_coreg (line 127)
In spm_run_coreg (line 22)
In cfg_run_cm (line 29)
In cfg_util>local_runcj (line 1688)
In cfg_util (line 959)
In spm_jobman>fill_run_job (line 469)
In spm_jobman (line 247)
In mri_coregister (line 156)
In import_mri (line 290)
In bst_call (line 28)
In tree_callbacks>@(h,ev)bst_call(@import_mri,iSubject,[],[],1) (line 562) 
Completed                               :          16:01:36 - 05/07/2021

SPM12: spm_reslice (v7141)                         16:01:36 - 05/07/2021
========================================================================
Completed                               :          16:01:41 - 05/07/2021
Done    'Coregister: Estimate & Reslice'
Done

in_mri_nii : No such file or directory
in_mri_nii : Error opening header file

***************************************************************************
** Error: Line 239: Reference to non-existent field 'Cube'.
** 
** Call stack:
** >mri_coregister.m at 239
** >import_mri.m at 290
** >bst_call.m at 28
** >tree_callbacks.m>@(h,ev)bst_call(@import_mri,iSubject,[],[],1) at 562
** 
***************************************************************************

BST> Plugin spm12 already loaded: G:\My Drive\NeuroTeeModeling\roast-master\lib\spm12
BST> Emptying temporary directory...
Item estwrite: No field(s) named
woptions

------------------------------------------------------------------------
Running job #1
------------------------------------------------------------------------
Running 'Coregister: Estimate & Reslice'

SPM12: spm_coreg (v6435)                           16:03:25 - 05/07/2021
========================================================================
Completed                               :          16:03:40 - 05/07/2021

SPM12: spm_reslice (v7141)                         16:03:40 - 05/07/2021
========================================================================
Completed                               :          16:03:45 - 05/07/2021
Done    'Coregister: Estimate & Reslice'
Done

in_mri_nii : No such file or directory
in_mri_nii : Error opening header file

***************************************************************************
** Error: Line 239: Reference to non-existent field 'Cube'.
** 
** Call stack:
** >mri_coregister.m at 239
** >import_mri.m at 290
** >bst_call.m at 28
** >tree_callbacks.m>@(h,ev)bst_call(@import_mri,iSubject,[],[],1) at 562
** 
***************************************************************************

>> `

ftadel, 2021-07-06:
Update SPM, your version looks very old!
(remove it from your path, and install it as a Brainstorm plugin)
And update Brainstorm: plenty of things where posted in June.

You should get something like this:

```
BST> Checking latest online version for spm12...
BST> Plugin spm12 already loaded: C:\Users\franc\.brainstorm\plugins\spm12
BST> Emptying temporary directory...
Item estwrite: No field(s) named
woptions

------------------------------------------------------------------------
06-Jul-2021 11:33:59 - Running job #1
------------------------------------------------------------------------
06-Jul-2021 11:33:59 - Running 'Coregister: Estimate & Reslice'

SPM12: spm_coreg (v7320)                           11:33:59 - 06/07/2021
========================================================================
Completed                               :          11:34:17 - 06/07/2021

SPM12: spm_reslice (v7141)                         11:34:17 - 06/07/2021
========================================================================
Completed                               :          11:34:24 - 06/07/2021
06-Jul-2021 11:34:24 - Done    'Coregister: Estimate & Reslice'
06-Jul-2021 11:34:24 - Done
```

ftadel, 2021-07-06:
I added a test to capture SPM errors:
https://github.com/brainstorm-tools/brainstorm3/commit/8162c25a3ac60cd17948e60c880a62b0080ca0fc

tmedani, 2021-07-07:
Thanks, indeed my spm version is old. 
I updated it and it works fine. 

Thanks

closed 2021-07-07 by tmedani