brainstorm-tools/brainstorm3 #426: Cat12 segmentation 5 tissue values

AlexBo924, 2021-07-20:
Good afternoon Francois,

I recently updated brainstorm and cat12 to the latest version (Version  1844 CAT12.8-Beta 2021-06-03) and having some issues with the voronoi calculation used in the nirstorm plugin.

The 5tissues segmentation obtained from CAT12 seems "not compatible".

The values on the legend go from 0 to 250 which I don't understand why, even if each segmentation is correct (here CSF = 3). Normally the legend should go from 0 to 5

I tried using an old version of the 5tissues segmentation and it works (legend goes from 0 to 5, which is what I expect). 

Thank you in advance

Alex
[image]

ftadel, 2021-07-21:
> The values on the legend go from 0 to 250 which I don't understand why, even if each segmentation is correct (here CSF = 3). Normally the legend should go from 0 to 5

Don't worry about the color bar on the right of the figure, I don't think it is used here.
It it were, it would be for the T1 MRI volume in the background of the tissue segmentation.
If the text in the top-right corner is correct (here: "CSF") is correct, I think this is what matters.

> I recently updated brainstorm and cat12 to the latest version (Version 1844 CAT12.8-Beta 2021-06-03) and having some issues with the voronoi calculation used in the nirstorm plugin.
The 5tissues segmentation obtained from CAT12 seems "not compatible".

Please post this question on the Brainstorm user forum, mention @tvincent @edoudela12, and include a copy-paste of the full error messages you get in the Matlab command window:
https://neuroimage.usc.edu/forums/

@thomas-vincent @Edouard2laire  FYI

ftadel, 2021-09-19:
@AlexBo924 Is this still an active issue?

Edouard2laire, 2021-11-05:
i think the issue got solved. Can you confirm @AlexBo924 ? :)

ftadel, 2022-04-18:
Closing this for now.
@AlexBo924  Please re-open it if you have anything to add.

closed 2022-04-18 by ftadel