brainstorm-tools/brainstorm3 #439: OpenMEEG migration heads up

papadop, 2021-09-20:
This is just a head's up message for now.

The old INRIA forge (gforge) will close soon (they say us the end of the month, probably to push for the migration).
Consequently, we will have to migrate the OpenMEEG binary files that brainstorm uses to a new place called files.inria.fr.

Consequently, the web adress "http://openmeeg.gforge.inria.fr/download/" mentioned several times in brainstorm sources will have to change sooner or later (and the sooner, the better). I will try to ask for a redirection so that not to break installed brainstorms. 

I will communicate the new URL here as soon as I will obtain the new one and have copied files from the old repository to the new one.

Sorry for the disturbance....

ftadel, 2021-09-21:
Thank you for the notice.
Please keep us posted when you have the new download URLs.

ftadel, 2021-10-01:
The download URLs are not valid anymore, there are no redirections set up, and we didn't receive the updated URLs yet...
Can you please help us with that?

ftadel, 2021-10-01:
https://neuroimage.usc.edu/forums/t/where-to-get-openmeeg-precompiled-binaries-problems-installing-it-from-matlab/30869/2

papadop, 2021-10-01:
Normally, things should now work with the URL: 
https://files.inria.fr/OpenMEEG/download/
This can replace the old URL in the brainstorm code. It will not break old code because there is a redirection from the old URL to the new one.

ftadel, 2021-10-01:
> Normally, things should now work with the URL:

URLs updated: https://github.com/brainstorm-tools/brainstorm3/commit/2f85368f32e38182964adf594e8ccdea6d01b107

> It will not break old code because there is a redirection from the old URL to the new one.

http://openmeeg.gforge.inria.fr/download/OpenMEEG-2.4.1-Linux.tar.gz redirects to https://openmeeg.github.io/
Not to: https://files.inria.fr/OpenMEEG/download/OpenMEEG-2.4.1-Linux.tar.gz

Therefore it does break the existing code...
Am I missing something?

papadop, 2021-10-01:
I have passed the information to our administrators...
Unfortunately, I do not have the hand on the problem.

agramfort, 2021-10-02:
thanks @ftadel for having updated the URLs in the source files 🙏

ftadel, 2021-11-07:
Ok, people will have to update Brainstorm before updating OpenMEEG.
That works too...

closed 2021-11-07 by ftadel