brainstorm-tools/brainstorm3 #443: Optimize compiled distribution of iso2mesh in compiled

ftadel, 2021-09-29:


fangq, 2022-05-24:
@ftadel, I am not sure I understood the "DeleteFiles" list in https://github.com/brainstorm-tools/brainstorm3/commit/f73b204998869274e292cb01d45899d66afc32e0, I see `cgalmesh.exe` is listed both in your above screenshot and the DeleteFiles, is it supposed to be removed?

I want also to mention that if your package does not intend to support 32bit windows, you can also remove all `.exe` when there is a `_x86-64.exe` available.

yes, all `.mexarmhf` should be removed as they are not commonly used (built for RaspberryPi).

another way to squeeze more space is to build these tools using the source code under `iso2mesh/tools`, this will apply dynamic linking to produce small files, but it will require your system to have some libraries pre-installed, such as libCGAL.

ftadel, 2022-05-25:
> @ftadel, I am not sure I understood the "DeleteFiles" list in https://github.com/brainstorm-tools/brainstorm3/commit/f73b204998869274e292cb01d45899d66afc32e0, I see cgalmesh.exe is listed both in your above screenshot and the DeleteFiles, is it supposed to be removed?

Brainstorm first downloads the .zip package defined in `bst_plugin.m`, then deletes the unnecessary files, listed in this variable `DeleteFiles`.  The screen capture above is old, it was made before the removal of the 32bit windows binaries. `cgalmesh.exe` is now removed.

The binaries that are kept are now:
[image]

> I want also to mention that if your package does not intend to support 32bit windows, you can also remove all .exe when there is a _x86-64.exe available.
> yes, all .mexarmhf should be removed as they are not commonly used (built for RaspberryPi).

Done.

> another way to squeeze more space is to build these tools using the source code under iso2mesh/tools, this will apply dynamic linking to produce small files, but it will require your system to have some libraries pre-installed, such as libCGAL.

I won't go there, the idea is to have something as portable as possible. 
One reason for which many people use Brainstorm is because it's easy to make it work everywhere, with no dependency headaches :)

I'm already very happy with these 10Mb saved.
Thanks!