brainstorm-tools/brainstorm3 #464: [MatlabProcessManager] Launching Freesurfer in the background

Edouard2laire, 2021-11-25:
Hello,

This is a comment for a potential future PR. I really like being able to call FS from brainstorm; however, the fact that we can't use Brainstorm while FS is computing makes it unpractical. In that case, it's easier to lunch FS from a terminal in the background.  

A solution could be to use a process manager such as https://github.com/brian-lau/MatlabProcessManager used for stan to have free surfer running in the background while still being able to continue using Brainstorm.  I am not opening a PR as there might be a lot of issues for example you start FS on a subject in Protocol A, then switch to Protocol B while FS is computing: it might create conflict. However; I the user is constrained enough, it would still allow analyzing some signal while FS is computing in the background. 

FOr exemple those lines (
https://github.com/brainstorm-tools/brainstorm3/blob/59a0de1346b6aa5f7ba7d41c4f27c1018c758f48/toolbox/process/functions/process_segment_freesurfer.m#L234-L236 ) : 

```
    bst_progress('text', ['<HTML>Running FreeSurfer recon-all ' strT1T2 '...<BR><FONT COLOR="#707070"><I>(see command window)</I></FONT>']);
    disp(['BST> System call: ' strCall]);
    status = system(strCall)
```

could be replaced by 

```
p = processManager('id','FS','command', strCall);
addlistener(p.state,'exit',@(src,data)import_anatomy_fs(iSubject, FsDir, nVertices, isInteractive, sFid, isExtraMaps, isVolumeAtlas, isKeepMri));
GlobalData.process = p;  % need to be saved
```

Usefull ressource: 
- https://github.com/brian-lau/MatlabProcessManager/wiki/Notification-when-a-process-finishes
- https://github.com/brian-lau/MatlabProcessManager/wiki/Running-processes-in-a-GUI

Let me know what you think, 
Edouard

ftadel, 2021-11-26:
Writing a proper job manager would be a good thing, it's actually in our future plans for being able to send jobs to other computation units. But we won't start working on this immediately, we have to wrap up our new database system first. 

For the moment, I added an option to run the FreeSurfer asynchronously, and then you need to import the anatomy manually:
https://github.com/brainstorm-tools/brainstorm3/commit/9f65c69668fc135987df052369fccd56aa735ed8

There is no tracking of the end of the process, but you can add "-mail username" to the extra command-line parameters:
https://surfer.nmr.mgh.harvard.edu/fswiki/recon-all

Does this help you, at least partially?

Edouard2laire, 2021-11-29:
> Writing a proper job manager would be a good thing, it's actually in our future plans for being able to send jobs to other computation units. But we won't start working on this immediately, we have to wrap up our new database system first.
> 
> For the moment, I added an option to run the FreeSurfer asynchronously, and then you need to import the anatomy manually: [9f65c69](https://github.com/brainstorm-tools/brainstorm3/commit/9f65c69668fc135987df052369fccd56aa735ed8)
> 
> There is no tracking of the end of the process, but you can add "-mail username" to the extra command-line parameters: https://surfer.nmr.mgh.harvard.edu/fswiki/recon-all
> 
> Does this help you, at least partially?

Nice. looking forward to the new database system :)

closed 2021-11-29 by ftadel