brainstorm-tools/brainstorm3 #469: BIDS for NIRS

Edouard2laire, 2021-12-03:
Hello, 

Recently, BIDS started to be implemented for NIRS (mainly using the snirf format, already supported by Brainstorm). I was wondering if you could guide me on what part of the code modify to import export BIDS dataset containing nirs data such as :  https://github.com/rob-luke/BIDS-NIRS-Tapping/tree/388d2cdc3ae831fc767e06d9b77298e9c5cd307b ?

Regards, 
Edouard

ftadel, 2021-12-03:
The two files to edit would be `process_import_bids.m` and `process_export_bids.m`.
- process_import_bids: 
   - line 541: add the the modality
   - line 608: add the file extension
   - line 676: add the sensor types
   - If there are new metadata files to import, this might require some more work, otherwise it could be pretty simple
- process_export_bids: I'm not familiar with it, I have never used it. This is only to export unmodified continuous data

Hopefully at some point, we will replace the parsing (import) and the file name generation (export) with methods called from BIDS-Matlab: https://github.com/bids-standard/bids-matlab

Edouard2laire, 2021-12-03:
Thanks. Started here: https://github.com/Edouard2laire/brainstorm3/tree/BIDS-nirs

So far i only have the following issue/ waning when importing: I will have a closer look monday

```
Warning: Column "filename" not found in file: /NAS/home/edelaire/Documents/data/BIDS-NIRS-Tapping/sub-01/sub-01_scans.tsv
BST> File converted to binary .bst format: /NAS/home/edelaire/Documents/bst_db/BIDS_NIRS/data/sub-01/@rawsub-01_task-tapping_nirs/sub-01_task-tapping_nirs.bst
Warning: Column "name" not found in file: /NAS/home/edelaire/Documents/data/BIDS-NIRS-Tapping/sub-01/nirs/sub-01_task-tapping_channels.tsv
Warning: Column "group" not found in file: /NAS/home/edelaire/Documents/data/BIDS-NIRS-Tapping/sub-01/nirs/sub-01_task-tapping_channels.tsv
Warning: Column "filename" not found in file: /NAS/home/edelaire/Documents/data/BIDS-NIRS-Tapping/sub-02/sub-02_scans.tsv
BST> File converted to binary .bst format: /NAS/home/edelaire/Documents/bst_db/BIDS_NIRS/data/sub-02/@rawsub-02_task-tapping_nirs/sub-02_task-tapping_nirs.bst
Warning: Column "name" not found in file: /NAS/home/edelaire/Documents/data/BIDS-NIRS-Tapping/sub-02/nirs/sub-02_task-tapping_channels.tsv
Warning: Column "group" not found in file: /NAS/home/edelaire/Documents/data/BIDS-NIRS-Tapping/sub-02/nirs/sub-02_task-tapping_channels.tsv
Warning: Column "filename" not found in file: /NAS/home/edelaire/Documents/data/BIDS-NIRS-Tapping/sub-03/sub-03_scans.tsv
BST> File converted to binary .bst format: /NAS/home/edelaire/Documents/bst_db/BIDS_NIRS/data/sub-03/@rawsub-03_task-tapping_nirs/sub-03_task-tapping_nirs.bst
Warning: Column "name" not found in file: /NAS/home/edelaire/Documents/data/BIDS-NIRS-Tapping/sub-03/nirs/sub-03_task-tapping_channels.tsv
Warning: Column "group" not found in file: /NAS/home/edelaire/Documents/data/BIDS-NIRS-Tapping/sub-03/nirs/sub-03_task-tapping_channels.tsv
Warning: Column "filename" not found in file: /NAS/home/edelaire/Documents/data/BIDS-NIRS-Tapping/sub-04/sub-04_scans.tsv
BST> File converted to binary .bst format: /NAS/home/edelaire/Documents/bst_db/BIDS_NIRS/data/sub-04/@rawsub-04_task-tapping_nirs/sub-04_task-tapping_nirs.bst
Warning: Column "name" not found in file: /NAS/home/edelaire/Documents/data/BIDS-NIRS-Tapping/sub-04/nirs/sub-04_task-tapping_channels.tsv
Warning: Column "group" not found in file: /NAS/home/edelaire/Documents/data/BIDS-NIRS-Tapping/sub-04/nirs/sub-04_task-tapping_channels.tsv
Warning: Column "filename" not found in file: /NAS/home/edelaire/Documents/data/BIDS-NIRS-Tapping/sub-05/sub-05_scans.tsv
BST> File converted to binary .bst format: /NAS/home/edelaire/Documents/bst_db/BIDS_NIRS/data/sub-05/@rawsub-05_task-tapping_nirs/sub-05_task-tapping_nirs.bst
Warning: Column "name" not found in file: /NAS/home/edelaire/Documents/data/BIDS-NIRS-Tapping/sub-05/nirs/sub-05_task-tapping_channels.tsv
Warning: Column "group" not found in file: /NAS/home/edelaire/Documents/data/BIDS-NIRS-Tapping/sub-05/nirs/sub-05_task-tapping_channels.tsv
```

ftadel, 2022-11-11:
Any update on this?

Edouard2laire, 2022-11-11:
Hello, 

I have been quite busy lately but I should be able to work on it soon. First I need to make a PR to improve a bit the snirf reader. then it will be mainly copying from the previous PR.

Edouard2laire, 2022-11-14:
ok. Done in #589. Can't believe already one year passed since we open this thread

commit 7b0128bf (2022-11-21): [BIDS] Import NIRS dataset (fixes #469) (#589)
closed 2022-11-21 by ftadel