brainstorm-tools/brainstorm3 #474: Loading FieldTrip without overriding Matlab functions

ftadel, 2021-12-16:
From Brainstorm, FieldTrip is initialized by calling `ft_defaults.m`.
https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/core/bst_plugin.m#L445

However, @rcassani discovered that by default this adds to the path some `external` subfolders that shadow Matlab functions.
Example: after running `ft_defaults`, the function `filtfilt` points at `fieldtrip/external/signal/filtfilt.m` instead of the implementation of the Matlab Signal Processing Toolbox.

This is probably not what we want to do, as it impacts other functions in Brainstorm. For example: we don't want the decoding processes to be behave differently depending on whether the FieldTrip is loaded as a plugin or not.

We can disable this behavior by calling `ft_default.toolbox.signal='matlab'; ft_defaults;` instead.

@robertoostenveld @schoffelen 
1) Is there any particular reason why you decide to use the FieldTrip implementations instead of the native ones?
2) Can we safely expect that all the shadowed functions are exact replacements for the Matlab versions? (all the combinations of parameters are supported)
3) If we set `ft_default.toolbox.signal='matlab'`, but the Signal Processing Toolbox is not available on the computer: does FieldTrip default it back to the FieldTrip implementation, or do we have to test for this beforehand?
4) What is the exhaustive list of toolboxes that we should prevent FieldTrip from overloading?

Thanks!

robertoostenveld, 2021-12-17:
For reference, this is (partially) documented on https://www.fieldtriptoolbox.org/faq/requirements/#external-requirements

Re 1: The reason for these external drop-in replacements and the default being `compat` rather than `matlab` is that at the Donders (and Radboud) we have a shared network license server with more MATLAB licenses than licenses for these individual toolboxes.  The widespread use of FieldTrip - especially on our compute cluster - caused us to run out of licenses and hence resulted in grumpy non-neuroscience colleagues.  

Re 2: The external signal, stats and images code is not really from FieldTrip (hence in external), but mainly from Octave. See also this https://github.com/fieldtrip/fieldtrip/issues/1941 which was filed 3 days ago. It might touch upon the same underlying cause, i.e. someone noticing that (after many years) something does not work any more. The issue triggered by Guillaume should result in us writing a test script that ensures compat/matlab code compatibility.    

Re 3: No, as that would still cause us to eat up all of the university licenses before we switch over (and others would be out of luck).

Re 4: signal, stats and images, see [here](https://github.com/fieldtrip/fieldtrip/blob/42b1aea488a90a777fe957b64823fd55d1b9b742/ft_defaults.m#L226) and in the documentation of `ft_defaults`.

ftadel, 2021-12-20:
Thanks for the explanations!

I picked the code from this page: 
https://www.fieldtriptoolbox.org/faq/can_i_prevent_external_toolboxes_from_being_added_to_my_matlab_path/

I added tests and now use `'matlab'` whenever possible:
https://github.com/brainstorm-tools/brainstorm3/commit/cbf2d1522db22a3236e062cb37e70583275c0c1d

```
global ft_default; 
ft_default = [];
if exist('filtfilt', 'file''), ft_default.toolbox.signal='matlab'; end;
if exist('nansum'', 'file'), ft_default.toolbox.stats='matlab'; end;
if exist('rgb2hsv', 'file'), ft_default.toolbox.images='matlab'; end;
ft_defaults;
```

I'm suspicious about `rgb2hsv` though, which is available in the general Matlab distribution when the Image Processing Toolbox is not installed. This one probably never needs a replacement. See: https://www.mathworks.com/help/matlab/ref/rgb2hsv.html

```
>> which rgb2hsv 
F:\Software\Installed\MATLAB_2013b\toolbox\matlab\specfun\rgb2hsv.m
```

ftadel, 2022-01-07:
@robertoostenveld @schoffelen
Can you explain what's the logic for `rgb2hsv.m`, which is never missing?
You could probably remove all the extra management related with the `Image processing toolbox`, right?

schoffelen, 2022-01-11:
hmmm, if I do which rgb2hsv on my Mac I get: /Applications/MATLAB_R2021a.app/toolbox/matlab/images/rgb2hsv.m

the directory toolbox/matlab/specfun does not contain an overloaded copy. So at first sight it looks somewhat version specific.

schoffelen, 2022-01-11:
I can confirm that matlab2013b on our linux cluster has indeed the rgb2hsv function in matlab/specfun. For matlab2021b on our cluster it's in images...

ftadel, 2022-01-12:
@schoffelen In the versions I tested, the `rgb2hsv.m` file is located in `images` when the Image processing toolbox is installed, and in the `specfun` folder when the toolbox is not installed. This function is never overloaded, its installation folder is different depending on the list of toolboxes installed. 
But the important point is that it's always avalaible, FieldTrip doesn't need any replacement for it.

schoffelen, 2022-01-12:
Hi @ftadel I typically tend to overestimate the users of our toolboxes, so if the function is not obstructing anything, I'd rather keep it in place.  Note that there is only a single function in external/images, and I would imagine that any other required function from the images toolbox is just taken from the corresponding matlab function, provided it's available. Note also that I have  PR (almost merged) that does a quick sanity check on the comparability between the matlab version and the overloaded functions: https://github.com/fieldtrip/fieldtrip/pull/1945

robertoostenveld, 2022-01-12:
The function being in `images` means that it would take a license from our university-wide concurrent network license server, where the number of image processing toolbox licenses is limited. Hence we have it in `fieldtrip/external/images`. If you don't want it there, don't add it to your path.

ftadel, 2022-01-13:
This all makes perfect sense.
I just wanted to make sure that this was done on purpose.

On our end, we'll always use the `matlab` option when possible, to make sure FieldTrip doesn't override any Matlab functions. We don't want to obtain different behaviors of Brainstorm or other plugins, depending on whether FieldTrip is in the path or not.

Thank you both for your explanations!

closed 2022-04-18 by ftadel