brainstorm-tools/brainstorm3 #499: Envelope correlation: Missing 1xN and NxN processes

ftadel, 2022-02-04:
The HENV functions developed by @HosseinShahabi still miss their generalization to any pair of signals.

At the moment, the function `bst_henv.m` is only capable of computing NxN graphs, as it only takes one matrix `Data` in input:
https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/connectivity/bst_henv.m#L1
 
It was requested from the beginning that this must work like all the other connectivity functions, with two lists of signals `X` and `Y`. Examples: 
- bst_cohn.m: https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/connectivity/bst_cohn.m
- bst_granger.m: https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/connectivity/bst_granger.m#L1

Implementing this requires some significant work, as many coding tricks won't apply.

Without this generalization, it is impossible to:
- Compute the envelope correlation between two different files (eg. one EMG sensor and the cortex, one scout and the rest of the brain)
- Compute 1xN measures without computing the full NxN graphs: which in practice makes it impossible to compute a brain map of HENV measures (1x15000) because it would require the computation of the full 15000x15000 graph.

These are standard use cases, therefore these restrictions that are too important for what we're expecting to provide. At the moment, we can't replace the old AEC processes with the new HENV processes because of this technical limitation.

--
Attached: the processes for 1xN and AxB computations, once `bst_henv.m` is made more generic:
[processes_henv.zip](https://github.com/brainstorm-tools/brainstorm3/files/8003591/processes_henv.zip)

HosseinShahabi, 2022-03-02:
Hi Francois, 

I have updated the code and now it accepts two different signals with a different number of channels. Would you please clarify what do you mean by trick? Like if x is (N,nX) and y is (N,nY) now you can compute the product with one line code (transposing one) and does not need for loop for each pair. Is there anything I'm missing here? 

Thanks,
Hossein

HosseinShahabi, 2022-03-02:
I am working on the Corr function to solve that as well, so no need to use octave functions.

HosseinShahabi, 2022-03-02:
Okay, would you please take a look at this? Only "oenv" remained to be updated. I will definitely check it with more simulated data but want to make sure it is what you are looking for. 

[bst_henv.zip](https://github.com/brainstorm-tools/brainstorm3/files/8166618/bst_henv.zip)

PR #523 (2022-05-30): Envelope correlation: Added 1xN and AxB processes
fixes #499
closed 2022-05-30 by ftadel