brainstorm-tools/brainstorm3 #528: Error importing FastSurfer folder

alx080, 2022-03-23:
Hello,
in your tutorial for [FastSurfer](https://neuroimage.usc.edu/brainstorm/Tutorials/SegFastSurfer) it is written that you can import the FastSurfer folder as illustrated in the FreeSurfer tutorial, but I get this error:

[image]

Is there a way to fix this?

ftadel, 2022-03-23:
Can you please share an example for us to reproduce the error?
- Duplicate the FastSurfer folder
- In the copy, delete all the files except needed by Brainstorm (listed here: https://neuroimage.usc.edu/brainstorm/Tutorials/LabelFreeSurfer#Files_imported)
- Make sure you still get the same error when trying to import this folder (make sure it's the same error message, not simply an error). If you get a different error: Keep the entire initial folder. It was just to save some space in the file transfer...
- Zip the folder
- Upload the .zip file somewhere
- Post the download link here
- 
Thanks

alx080, 2022-03-24:
It was the same error, also this time I used the example MRI from FastSurfer's [Google Colab](https://colab.research.google.com/github/Deep-MI/FastSurfer/blob/master/Tutorial/Complete_FastSurfer_Tutorial.ipynb).
Here is the folder: [Link](https://drive.google.com/file/d/1D-BKy2LO2ASgJ2HyqXTRe24-DWSkco8w/view?usp=sharing)

ftadel, 2022-03-25:
The files `lh.white` and `rh.white` are empty: something went wrong either during the FastSurfer computation, or during the copy of the results (e.g. you copied the output folder with a tool that can't handle symbolic links).

Two solutions:
- You fix the `white` surfaces from the input folder
- You simply delete them (everything else will work):
[image]
[image]

closed 2022-03-25 by ftadel