brainstorm-tools/brainstorm3 #536: Duplicate events process

guiomar, 2022-04-28:
Hi! 

Is there an option to duplicate events through a process using the Run button? 
I see there is the option in the events panel_record, and most of these options can also be acessed through a process using the Run button, but not the duplicate. What would be the most efficient way of doing it?

rcassani, 2022-04-28:
Hi @guiomar, it is possible to use the Merge events process (`process_evt_merge.m`) to this end, with the following parameters:

- **Events to merge**: `event-to-copy, event-to-copy`
- **New event name**: `copy-of-event`
- Uncheck **Delete the original event**

I agree, it seems better to have specific process for this. @ftadel?

ftadel, 2022-04-29:
Process "Merge events" transformed into "Duplicate / merge": https://github.com/brainstorm-tools/brainstorm3/commit/0e87b06fdc97278f878e8eafaebe0a5625021d98