brainstorm-tools/brainstorm3 #541: creating new protocol breaks on ubuntu

lukeinator42, 2022-05-09:
In the latest versions of brainstorm, creating a new protocol crashes. My operating system is ubuntu 20.04, and this error occurred on matlab 2021a as well as matlab 2022a. 

I checked out previous commits to see if this issue still occurred and it looks like 3f20be8ff8516c24873552c45bb7b7fce8247b12 was the last version that still works.

The error message it throws is:
_Index exceeds the number of array elements. Index must not exceed 0.

Error in db_parse_subject (line 187)
if (length(sSubject(1).Anatomy) >= 2)

Error in db_load_subjects (line 125)
ProtocolSubjects.Subject = db_parse_subject(ProtocolInfo.SUBJECTS, '', 45);

Error in db_reload_database (line 74)
    db_load_subjects(0, isFix);

Error in db_load_protocol (line 93)
        db_reload_database(iProtocols(i));

Error in gui_brainstorm>SetCurrentProtocol (line 1046)
        db_load_protocol(iProtocol);

Error in gui_brainstorm (line 33)
eval(macro_method);

Error in gui_edit_protocol/updateProtocolModificiations (line 180)
        gui_brainstorm('SetCurrentProtocol', Protocol);_

rcassani, 2022-05-09:
Hi @lukeinator42, thanks for pointing out the bug. This will be promptly solved.

closed 2022-05-10 by ftadel

ftadel, 2022-05-10:
I commit buggy code in the evening, and 4hrs later it's already fixed...
You guys are amazing! Thanks :)

PR #542 (2022-05-10): Bugfix: Skip sort anatomy when no subjects
This PR aims to close #541
commit a169a178 (2022-05-10): Bugfix: Skip sort anatomy when no subjects. Closes #541 (#542)