brainstorm-tools/brainstorm3 #543: Surface registration after resampling

Edouard2laire, 2022-05-11:
When resampling an MRI, the coregistration between the resampled MRI and the surface extracted from the original MRI is lost. I think it might be expected but it might be good to display a warning / and or to remove the surface.

ftadel, 2022-05-12:
This is not expected.
Resampling/resizing the MRI should correctly update the coordinates of all the anatomical fiducials (NAS, LPA, RPA).
=> To check this: right-click on the resampled MRI > Edit MRI > Yes.
The coregistration is based on these three points, and if their position is correct, the coregistration with all the 3D points (surfaces, channel files...) should be maintained.

Can you please describe a way to reproduce this error?
Maybe starting from the introduction tutorials dataset, for easy reproducibility?
Explain click by click, from the import of the FreeSurfer folder, how to illustrate this "loss of coregistration".

Edouard2laire, 2022-05-14:
Hello, 
Thanks for your reply. Interestingly, the fiducials seems to be at the correct location even after the resampling but if i overlay the resampled MRI with the original one, they don't align : 

[image]

Interestingly, it seems that resclicing the MRI after the resampling (resclice > subject coordinate) is helping with the issue but i am not sure at all why. 

To reproduce; 
- Create a subject and use MNI > ICBM512 template
-  Right click on MRI, resample MRI with the following parameters;
-  MRI size: 197, 233,189 ; MRI resolution: 2,2,2; method: linear 

Here is an interesting stuff: 
- If you right click on the new MRI, and set it as default, then right click on cortex, and use MRI registration > check MRI/surface registration, they are aligned then if you put back the orignal MRI as default and do the same procedure, they are not aligned

- Another interesting fact, if you do the procedure in reverse order first check the coregistration between the surface and orignal MRI and then between surface and resampled MRI then it is the resampled MRI that is not aligned. 

I think it might be due to the fact the the interpolation matrix is not recomputed when changing the default MRI ?
 

While writing that

ftadel, 2022-05-17:
> edit: for the overlay of the two MRI mentioned at the beginning of this message, i think it's mainly visual as i guess that the MRI viewer is assuming that the voxel size of the two MRI is the same when creating the overlay

Exactly, the MRI viewer overlays voxel by voxel, ignoring the voxel size.
I committed a fix to display an error when the resolutions are not matching: https://github.com/brainstorm-tools/brainstorm3/commit/1e42e03806388ddcee51f584c56965db23b5d031

> I think it might be due to the fact the the interpolation matrix (tess2mri_interp) is not recomputed when changing the default MRI ? 

Indeed. I fixed this too: https://github.com/brainstorm-tools/brainstorm3/commit/53696f1a9878315d9b2e33ee986dddc8572283c5

Thanks for the suggestions!

closed 2022-05-17 by ftadel