brainstorm-tools/brainstorm3 #571: Connectivity: wPLI and trial-averaging

Moo-Marc, 2022-09-07:
After reviewing the math and connectivity code to answer a [forum question](https://neuroimage.usc.edu/forums/t/some-questions-about-resting-state-eeg-source-localization/36646/3), I flagged some issues in the recently added wPLI measure (PR https://github.com/brainstorm-tools/brainstorm3/pull/497, related [forum post](https://neuroimage.usc.edu/forums/t/weighted-phase-lag-index/9434)), and in trial-averaging that may apply to other measures as well.

1. I don't think the code actually corresponds to wPLI as it was published. It uses phase differences, phaseA\*phaseB', whereas wPLI in Vinck et al. 2011 is defined with HA\*HB'. @danielemarinazzo Where did this formula come from?
2. Furthermore, it's not currently the "debiased" version. So [this comment](https://github.com/brainstorm-tools/brainstorm3/blob/8ffcfd6eb7339116fefdfed699aabd5561af1682/toolbox/connectivity/bst_connectivity.m#L599) should be corrected.
3. There's also an important question of how trial-averaging should be performed (which we can discuss at a Brainstorm meeting). Many of the connectivity measures, such as wPLI, have multiple expectation value terms, or other operations after the expectation value. This means the computation should proceed in two steps: trial averaging each expectation value term separately, then the final assembly of the terms after looping through files (like how the absolute value is currently applied for PLV after looping, which is currently done in the [SaveFile](https://github.com/brainstorm-tools/brainstorm3/blob/8ffcfd6eb7339116fefdfed699aabd5561af1682/toolbox/connectivity/bst_connectivity.m#L864) subfunction). Computing the measure for each trial and then averaging is not equivalent and will probably result in increased variance among other things. This is similar to what was done for coherence (in a somewhat roundabout way). This would also allow implementing the debiased wPLI, "unbiased square PLV", etc.

danielemarinazzo, 2022-09-07:
Hi @Moo-Marc 

1. it's simple math, check this code

```
clear
load('HA_HB_iA_iB.mat') % any HA HB iA iB combination, [here](https://www.dropbox.com/s/pwuxmchfs62v0j8/HA_HB_iA_iB.mat?dl=0) an example
[nA,~]=size(HA);
[nB,ntime]=size(HB);
R = reshape(mean(sin(angle(HA(iA,:)')-angle(HB(iB,:)')))' ./ mean(abs(sin(angle(HA(iA,:)')-angle(HB(iB,:)'))))',[],nB);
R=abs(R);
phaseA = HA ./ abs(HA);
phaseB = HB ./ abs(HB);
num = abs(imag(phaseA*phaseB'));
den = zeros(nA,nB);
for t = 1:ntime
    den = den + abs(imag(phaseA(:,t) * phaseB(:,t)'));
end
wPLI_db = num./den;
c=corr(wPLI_db(:),abs(R(:)), 'rows','complete')
```

2. It is the debiased version, check this code

```
clear;clc
load('test.mat'); % example dataset [here](https://www.dropbox.com/s/jq7bozt5efu6rme/test.mat?dl=0)
%HA=HA(1,:);
[nA,ntime]=size(HA);
[nB,ntime]=size(HB);
% commented below, since we found out that repmat-ing HA directly is faster (see
% below)
%iA = repmat(1:nA, 1, nB)';
%iB = reshape(repmat(1:nB, nA, 1), [], 1);
R_wPLI=zeros(nA,nB);

% PLV
tic
phaseA = HA ./ abs(HA);
phaseB = HB ./ abs(HB);
csd=phaseA*phaseB';
R_PLV=abs(csd/ntime);
t=toc;
disp(['PLV, ' num2str(t) ' seconds']);

% ciPLV
tic
phaseA = HA ./ abs(HA);
phaseB = HB ./ abs(HB);
csd=phaseA*phaseB';
R_ciPLV=abs((imag((csd))/ntime)./sqrt(1-(real((csd))/ntime).^2));
t=toc;
disp(['ciPLV, ' num2str(t) ' seconds']);

% wPLI
tic
R_wPLI=zeros(nA,nB);
for itime=1:ntime
    phaseA_t = HA(:,itime) ./ abs(HA(:,itime));
    phaseB_t = HB(:,itime) ./ abs(HB(:,itime));
    csd=phaseA_t*phaseB_t';
    cdi = imag(csd);
    R_wPLI=R_wPLI+(abs(cdi).*sign(cdi))'./abs(cdi)';
end
wPLI=abs(R_wPLI/ntime);
t=toc;
disp(['wPLI, ' num2str(t) ' seconds']);

% wPLI debiased
tic
HA=HA';HB=HB';
sin_pd=sin(angle(repmat(HA,[1 nA])./repelem(HB,1, nA)));
a=abs(mean(sin_pd)./mean(abs(sin_pd)));
R_wPLI(1:length(a))=a;
t=toc;
disp(['wPLI phase difference, ' num2str(t) ' seconds']);

% faster implementation of wPLI (debiased), identical to the previous one
tic
num = abs(imag(phaseA*phaseB'));
den = zeros(nA,nB);
for t = 1:ntime
    den = den + abs(imag(phaseA(:,t) * phaseB(:,t)'));
end
wPLI_db = num./den;
t=toc;
disp(['wPLI db summing terms, ' num2str(t) ' seconds']);

if(any([nA,nB]==1))
    c=corr(wPLI_db(:),R_wPLI(:), 'rows','complete');
else
    c=compareconn(wPLI_db,R_wPLI);
end
disp(['comparison between the two wPLIs = ' num2str(c)])

if(any([nA,nB]==1))
    c=corr(wPLI_db(:),wPLI(:), 'rows','complete');
else
    c=compareconn(wPLI_db,wPLI);
end
disp(['comparison between wPLI db and wPLI = ' num2str(c)])

if(any([nA,nB]==1))
    c=corr(wPLI_db(:),R_ciPLV(:), 'rows','complete');
else
    c=compareconn(wPLI_db,R_ciPLV);
end
disp(['comparison between wPLI db and ciPLV = ' num2str(c)])
```
Also bottom line, in my opinion wPLI should be discarded and substituted by the much faster and basically equivalent ciPLV

3. Not sure it is relevant given the above.

Happy to discuss further!

danielemarinazzo, 2022-09-07:
I created a repo so it's neater

https://github.com/danielemarinazzo/test_brainstorm

danielemarinazzo, 2022-09-20:
Dear @Moo-Marc 

1. The code is just a faster one, the result is the same
2. You are right, I have now pushed a new version with the comment removed, and the debiased implementation commented below
3. Indeed the way Brainstorm hands the trials makes it difficult to implement the time-varying version of the debiased wPLI, we should check if it's possible to do it (maybe starting from the coherence routines).

For me, it's a reason more to use ciPLV instead.

danielemarinazzo, 2022-09-20:
This function
https://github.com/danielemarinazzo/test_brainstorm/blob/main/test_timevarying.m
implements the debiased wPLI (together with other measures).
The thing is that the way trials are handled now in Brainstorm makes it difficult (at least for me) to properly implement it.

Moo-Marc, 2022-09-20:
Thanks for checking this again Daniele.  I also spent some time yesterday reviewing the math and code and I was going to write a reply today.  So I think for 2 and 3 we are in agreement.  For 3, I'll propose a PR to do averaging properly and also implement the debiased version.

For 1 however, I stand by my earlier comment.  The paper (Vink et al. 2011) defines wPLI as a function of frequency:
```
CrossSpectrum = F1 * F2';  % where F1 is the Fourier transform of signal 1.
WPLI = abs(E{imag(CrossSpectrum)}) / E{abs(imag(CrossSpectrum));
```
Where E{.} is the expectation value, i.e. what needs to be averaged over trials.
The issue is that in all your formulae, you replaced the cross spectral density by (exp(i*Phase1) * exp(i*Phase2)') or something mathematically equivalent.  It's missing the amplitude part of the cross-spectrum.

What I verified yesterday (wasn't obvious to me), was how to go from the cross spectrum definition of WPLI, which is a function of frequency only, to a compatible extended definition in terms of instantaneous phase and envelope, a function time.  To get an equivalence, we can apply the same narrow band pass filter and integrate over frequencies in the first formula, and integrate over time (take the expectation value) in the second formula.  For the cross spectrum, this gives this relation:
[image]
where hat indicates Fourier transform, "a" subscript indicates analytical signal (Matlab hilbert), tilde indicates the signals were filtered with a narrow band pass, and W is the corresponding narrow band filter window in the frequency domain.  So the equivalent WPLI formula must replace CrossSpectrum by (H1 * H2'), the full analytical signals including their amplitudes, not just the exp(i*phase) part.  Those amplitudes are precisely the relative weights that give the name "weighted PLI".

Again, here the separate averaging of the E{.} terms is crucial, otherwise we could cancel some factors in the numerator and denominator.

I hope this clarifies what I meant.  Please let me know what you think.

danielemarinazzo, 2022-09-20:
Hi
I think you're right wrt to using HA and HB.
I have now implemented the fixes in the test repo https://github.com/danielemarinazzo/test_brainstorm
Now, for the static case it can be easily fixed (I have updated the PR), not sure though for the time-varying, where actually your point 3 makes even more sense now. Maybe wPLI should be implemented started from the coherence, as I suggested quite some time ago. 
Or maybe there's another easy solution I'm missing at the moment.
In either case I would suggest to hide wPLI for the moment. @ftadel any suggestion?