brainstorm-tools/brainstorm3 #586: Progress bar text on Linux displays previous text

Moo-Marc, 2022-10-21:
I noticed this a while ago.  (Just tried debugging but couldn't find a fix.)

The progress bar doesn't display the correct text inside the window when it appears. It shows the text from the previous time.  The correct text appears if we resize the progress bar window.  I tried drawnow, printing to the command window (given apparently similar bug in Linux with fix at line 155), setting the text after the window is visible, nothing worked.

Steps to reproduce: Put breakpoint in bst_progress line 223, start a process input files. The progress bar appears, resize it. It should show "reading input files...". Continue execution, the code pauses again at line 223.  The progress bar should now show "initialization...", but it's still "reading input files...".  Resize the window and the correct text appears again.

This is true on Linux (Ubuntu 18) Matlab 2021b.  If you can't reproduce, let me know what else to test; other Matlab versions, other possible fixes...

Cheers,
Marc

ftadel, 2022-10-25:
Unfortunately I can't reproduce this on my end.
My Ubuntu is running on the Win10/WSL, and this doesn't seem to happen.

Could you try assembling a single .m script that reproduces this problem?
- A script that requires brainstorm to be running but that does not use `bst_process` or any file in the database
- It would only make calls to `bst_progress` and random things to wait in between (either calls to `wait` or loops computing random things)
- This would help for reproducing the issue identically in different contexts, and find some code to force refreshing the label

For forcing to redraw the label, you could try adding Swing method calls on the various Java objects, either the text label or the frame (`pBar.jWindow`), the panel (`pBar.jPanel`), or the label (`pBar.jLabel`), immediately after setting the text (lines 219 and 285).
It could be functions like: `invalidate`, `revalidate`, `repaint`, `pack`, `setSize`, `setLocation`...
With the debugger stopped in the middle of `bst_progress`, you can see all the Java objects methods with `methodsview`, or by typing the name of object in the command window followed with a dot "." and pressing `TAB` to get the possible automatic completion.

You could also try various window managers on your Ubuntu system, as this is changing a lot the way the graphic objects interact and get refreshed.

rcassani, 2022-11-01:
Same here, I was not able to reproduce it.
Linux (Cinnamon DE) and Matlab 2020b