brainstorm-tools/brainstorm3 #607: BEM surfaces computation can be improved

ftadel, 2023-03-31:
I spent some time on trying to refine the BEM surface computation using all the new tools that are available.
The main improvement would be to start from the tissues segmentation volume when available.

At the moment, two solutions are available: "Brainstorm" and "FieldTrip".
https://neuroimage.usc.edu/brainstorm/Tutorials/TutBem#BEM_surfaces

**Brainstorm**: 
Inner skull surface from the default anatomy, warped in a weird way to the subject anatomy.
Two problems: Not based on the real subject anatomy, and the shape does not follow correctly the shape of the brain (flat between the poles of the frontal and temporal lobes).
[image]

**FieldTrip**:
Calling process_tf_volume_segment. Recomputes a very rough approximation of the tissues segmentation.
Two problems: Has to recompute a bad tissue segmentation even when a better one exists (CAT12 or SimNIBS), and the inner skull surface is not enclosing correctly the cortex surface (and therefore causes problems with OpenMEEG).
[image]

I started some developments to compute the inner skull surface based on the tissues segmentation from the subject anatomy (generated with SPM12, CAT12, SimNIBS or FieldTrip),  but could not finish them. I attach two functions below that both have problems. 

**tess_bem_tissues.m**:  Computation of the inner skull from the tissues segmentation
- Missing computation of outer skull: either by inflating the inner skull surface, or by dilating the inner skull mask (or even better: by computing the surface from the tissues at least for the top part of the head)
- Missing computation of the head surface
[tess_bem_tissues.m.txt](https://github.com/brainstorm-tools/brainstorm3/files/11123328/tess_bem_tissues.m.txt)

**tess_remesh_ray.m**: Attempt to replace tess_remesh.m with a ray tracing function.
- Problem: the ray tracing function misses too  many points and is not robust enough for this use
[tess_remesh_ray.m.txt](https://github.com/brainstorm-tools/brainstorm3/files/11123358/tess_remesh_ray.m.txt)

Edouard2laire, 2023-04-01:
Hello, 

Another option would be to use FSL BET : https://fsl.fmrib.ox.ac.uk/fsl/fslwiki/BET/UserGuide#bet I can send you a script next week if you want to try :) 

I am curious, do you know how much this change will will affect the forward model and the source localization ?

ftadel, 2023-04-03:
> Another option would be to use FSL BET

We already know how to extract the brain mask from a variety of tools (SPM, CAT, SimNIBS, FieldTrip).

The problem is to produce sets of BEM surfaces (innerk,outer,skin) that are compatible with OpenMEEG:
1) All three layers must be very smooth and regular (hence the approach to grow a sphere with ~2000 vertices to match the mask, both in Brainstorm and FieldTrip). It is impossible to use the realistic skull surfaces from SimNIBS or BrainSuite.
2) Ideally the outer skull should be a dilation of the inner skull, with vertices aligned across the two surfaces (c.f. @agramfort @papadop)
3) All the cortex surface vertices (= the dipoles) must be enclosed in the inner layer

> I am curious, do you know how much this change will will affect the forward model and the source localization ?

Not at all :/