brainstorm-tools/brainstorm3 #648: Integration of fMRI in brainstorm

Edouard2laire, 2023-10-11:
Hello, 
The aim of this thread is to discuss the potential of integrating fMRI data analysis into Brainstorm.  Having fMRI time course in the volumetric space would require too much modification to the existing code base of Brainstorm but I think we can integrate surface-based analysis quite easily. 

The main objective for now would be to be able to import minimally processed fMRI data on the cortical surface (typically the output from fMRIprep) to be able to then use existing functionally offered by Brainstorm/nirstorm for surface data analysis. 

The following post we be used to keep track of the progress :)

Edouard2laire, 2023-10-11:
Data importation
- [ ] Import derivate folder from fMRIprep
- [ ] Project 4D volume on the cortical surface using Voronoi ( https://github.com/Nirstorm/nirstorm/pull/213) - note that this function is currently part of NIRSTORM but could be moved to brainstorm if we want to include volume-to-surface projection in brainstorm.

One issue is that surfaces always need to be linked to a data file, notably for the importation of events. We would probably need to create a dummy data file/channel to keep the events functionalities.  From fMRIprep, we can import the confound file containing useful time course. 

Analysis 
- To write 

Misc
- [ ] Detect BOLD signal and don't flip the sign of the signal

rcassani, 2023-10-26:
That would be a good feature to have!

Regarding the data file, it does make sense to have it with confounds and to store events.  

Heads-up @tmedani