brainstorm-tools/brainstorm3 #650: [BIDS] import channel outside process_import_bids

Edouard2laire, 2023-10-22:
Hello

I was recently trying to import a channel formatted as BIIDS (optodes.tsv + sidecard json) on nirs file already imported into the database. 

It seems that if we import everything at once using "import dataset as BIDS", it is working but if we only import the channel coordinate, then the sidecar json file isn't read and therefore, it  is missing crucial information for the importation such as
- the coordinate system, - the unit of the coordinate, and most importantly, the fiducial coordinate. 

Regards, 
Edouard

rcassani, 2023-10-23:
Hi,

That's right, Under the hood, when importing the entire (NIRS) BIDS dataset, first the channel file is crate with the channel information obtained from the `snirf` file with `in_data_snirf`. After that, the optode locations are updated with the locations in the `_optodes.tsv`, using the sidecar to know the proper `SPACE` and `UNIT`.

If the NIRS is already in the database, it should have a channel file (created when reviewing the raw data), and only the locations will be updated with the `_optodes.tsv` file. This is not done as importing channel, but with the menu `Add EEG positions > Import from file` (yup). As only one file is provided when importing a channel, the user need to select in the `file of type` format for the expected `SPACE` and `UNITS` as `EEG:BIDS-SPACE-UNITS`.

All the `SPACE` options are available in the GUI, but only millimeter `UNITS` (MM).
If `UNITS` are centimeters or meters, a script can be used, see the link below.
💡 In the script the tag to indicates BIDS, `SPACE` and `UNITS` does not have the string `EEG` in it

(By coincidence a related topic was [asked in the Brainstorm forum today](https://neuroimage.usc.edu/forums/t/process-import-channel-not-working/42735/2)

Edouard2laire, 2023-11-11:
Thanks a lot for your reply. I agree that we can specify the space and unit in the GUI. but I am more worried about the specification of the fiducial points used for the registration that are only present in the sidecar file.