brainstorm-tools/brainstorm3 #657: Loading old database

Edouard2laire, 2023-11-15:
Hello, 

I am contacting you as I am trying to load an old database from a colleague. However, this database was created with an old version of Brainstorm and when I try to load the map I get the following error because sSubject.iFibers and sSubject.iFEM don't exist. WOuld there be an easy way to fix that and load the maps? 

[image]

Thanks
Edouard

rcassani, 2023-11-15:
Have you tried already to regenerate the metadata of the problematic protocol (`protocol.mat`) from the files in HDD?

https://neuroimage.usc.edu/forums/t/brainstorm-database-doesnt-load/36808/2

Edouard2laire, 2023-11-18:
Ohh it's actually a message I posted haha. I'll try and reopen if it doesn't work :)

closed 2023-11-18 by Edouard2laire