brainstorm-tools/brainstorm3 #674: Integration of multivariate connectivity methods

robbisg, 2024-01-09:
Hi, 

we are trying to implement a couple of multivariate connectivity methods Multivariate Interaction Measure (MIM) and Multivariate Phase Slope Index (MPSI) to then integrate them with BST.
We created all the files needed such as process_multivariate.m and modified the bst_connectivity to cope with these new metrics. 
We are facing some issues in the management of M-scouts since it seems that the scout functions implemented in BST allow only the reduction of the (3x)NxT sources to a 1xT array using first PC or mean or median, are we right?
We would like to obtain the Mx(3xN)xT set of scout timecourses and then apply for example a PCA using the first k-components., in order to run our methods in the M x k x T. Is it possible? How? Thanks a lot
Cheers,
Roberto

Moo-Marc, 2024-01-10:
Hello,

I'll give a quick answer, but this would be best discussed through the Brainstorm forum.

Short answer: it's not currently available.  You could edit the Brainstorm code to have this option, but it may require some effort.

When working on PCA for scouts and for connectivity last year, I tested at some point saving the first 3 components.  But we decided to just keep the first.  Since there were more development afterwards, I can't easily recover that code to share with you in a usable state.  You can have a look at the pcag3 option in this commit to get an idea of how it could be implemented.
https://github.com/brainstorm-tools/brainstorm3/pull/559/commits/e3684a37ba203e99525891a2b09fef24c4a2acb7

Hope it helps!
Cheers,
Marc

robbisg, 2024-01-11:
Thanks, I'll post on the forum to better discuss! :)