brainstorm-tools/brainstorm3 #678: Question about connectivity computing with OutputMode 'concat'

DerfelLee, 2024-01-25:
I think there's a mistake in the code about 'concat' but I'm not sure. The variable 'FilesA' is changed to 'FilesA(1)' but the variable 'nFiles', which should be changed to 1, stays the same with no changes. Therefore, the program loops 'nFiles' times calculating the same data over and over again.

[image]

But I'm not sure about this. Please tell me if I'm making a mistake.

Thanks!

Moo-Marc, 2024-01-25:
I confirm this is a bug.  The solution nFiles = 1 is acceptable, though we might want to keep the number of input files for use in some messages.

rcassani, 2024-01-30:
Bug fixed in [4144f44](https://github.com/brainstorm-tools/brainstorm3/commit/4144f4416c21165579a90feb5507b57aec2263bb)

closed 2024-01-30 by rcassani