brainstorm-tools/brainstorm3 #691: add time offset not shifting events time

ShahlaBakian, 2024-03-05:
Hello

we have a signal we cut using "extract time" now we want the signal to be started from 0 sec. so we used preprocess>add time offset. this is working but is removing all the events from the file. after investigating it seems like “prcocess_time_offset.m” is not shifting the event.

best regards,
Shahla

rcassani, 2024-03-05:
@ShahlaBakian, thank you for the heads-up
The bug is fixed in [65c4935](https://github.com/brainstorm-tools/brainstorm3/commit/65c4935720b992e2ca2f6a539dc59616ee0692a2)

Update your Brainstorm instance to get the fix.

P.S. Besides GitHub, you can post your questions, requests and bug reports in the Brainstorm forum:
https://neuroimage.usc.edu/forums/

closed 2024-03-05 by rcassani

ShahlaBakian, 2024-03-06:
Thanks alot Raymundo.