brainstorm-tools/brainstorm3 #730: add support to export the bst surfaces as STL files.

tmedani, 2024-07-25:
hi guys, 

it seems STL is one of the most used and common format for the surfaces meshes. 

currently, BST has these format: 
[image]

This issue is just to remind the need to add the STL file format to the export function.

Edouard2laire, 2024-07-26:
> This issue is just to remind the need to add the STL file format to the export function.

As a temporary workaround until Brainstorm can export directly STL file; it is possible to use freesurfer to convert gii file to STL using mris_convert

rcassani, 2024-07-26:
To create `.stl` files from Brainstorm surfaces, I export them as either  `.off` or `.obj`, then use [MeshLab](https://www.meshlab.net/).

closed 2025-02-19 by rcassani